lemmingDev/ESP32-BLE-Gamepad #165: How to send an int value to ESP32 as gamepad from Python or C program

AlEnterni, 2023-03-15:
I'd like to ask  if there's a way to send an int value from a Python or C program to the ESP32, while it's working as a gamepad. Specifically, I want to know if there's a method for achieving this using the existing codebase or if I would need to modify it. Any guidance on how to accomplish this would be much appreciated.

baohuiming, 2023-03-22:
Not sure if I understand you correctly. Maybe you need to create a Bluetooth client to communicate. You can refer to [NimBLE-Arduino](https://github.com/h2zero/NimBLE-Arduino/tree/release/1.4/examples)

lemmingDev, 2023-03-25:
Try this version

It allows you to send and receive as much info as you live over a BT Classic serial connection while using the gamepad

https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/111#issuecomment-1254260699

AlEnterni, 2023-03-26:
Would I need to connect to the ESP32 twice then? One for it to work as a gamepad and another for data? 
Thank you very much!

LeeNX, 2024-11-23:
I was able to put together a small sketch that has NUS, which is basically BLE Serial, that could be used - https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/195#issuecomment-2488799340
  
Or you could also try a PR - https://github.com/lemmingDev/ESP32-BLE-Gamepad/pull/234
That adds ```Output Report Function```, meaning you can send data from host OS to gamepad.  
  
Would be best if the feature was general enough, that the game engine or game building library does not need customisation.

LeeNX, 2026-08-25:
Hey @AlEnterni , checking in if you came right?

There are now a few ways you can get data in/out of the gamepad, ie: BLE NUS Serial, ```Output Report Function``` or a new PR #283 which can be both ways. There are examples of test each of these options.

Please let us know if we can close this issue, thanks.

LeeNX, 2026-08-27:
PR merged and a basic sketch and python testing application that could be used is in place, take a look at [TestFeatureReports](https://github.com/lemmingDev/ESP32-BLE-Gamepad/tree/master/examples/TestFeatureReports)

I should add a walkthrough on how to setup and use.