lemmingDev/ESP32-BLE-Gamepad #170: bleGamepad.setRightThumb(x,y);

bihailantian655, 2023-04-15:
#bleGamepad.setLeftThumb(a,b);   working fine ；

 bleGamepad.setRightThumb(x,y);   This function is not working properly.

It was supposed to control Axis3, but it actually controls Axis5, just like the picture below;

[image]

bihailantian655, 2023-04-15:
bleGamepad.setZ();
      bleGamepad.setRX();

These functions can be used instead for the time being

NopeProsOnly, 2024-11-07:
This is still an issue, but it looks like a relatively easy fix. If it doesn't get fixed soon, I might even send in a PR.

lemmingDev, 2025-02-01:
Has been fixed in latest master

Will push out a new release soon

Thanks

closed 2025-02-01 by lemmingDev