lemmingDev/ESP32-BLE-Gamepad #171: compatibility with BLE-Mouse

vale-max, 2023-04-17:
HI LemmingDev,
I self-built a simil-arcade Joystick using an ESP32 and ESP32-BLE-Keyboard library, now I switched to your gamepad library and it's better as it is seen as a proper controller.
There's a side issue though.
I used the same joystick as a mouse emulator to keep a pc from waking up. By keeping a button pressed at power-on the thing entered mouse mode using ESP32-BLE-Mouse and just sent some random movement every now and then.

This does not work with gamepad library as the functions in BleConnectionStatus.cpp are duplicated between BleMouse and BleGamepad libraries, causing compilation to fail.

Is it possible to change the names of these functions in order to be able to include both libs and be able to decide at boot whether to work as gamepad or mouse?

lanshicai, 2023-04-24:
I also want to use the game controller and Bluetooth keyboard at the same time.

LeeNX, 2023-05-05:
I don't believe this is an easy ask. Both libraries ( and others, like the keyboard library ) would need to be merged into a single library, a little like https://github.com/blackketter/ESP32-BLE-Combo.

closed 2024-02-24 by lemmingDev