lemmingDev/ESP32-BLE-Gamepad #175: Proper credits

douggmill, 2023-05-09:
I recently made a report demonstrating a Lego arcade joystick I built, I used the BLE-Gamepad library and posted the code I used on my Github. Here is a link to the pdf report: 

https://github.com/douggmill/BLE-Arcade-Joystick-LEGO-Arduino/blob/master/LEGO%20Arduino%20Arcade%20Joystick%20Home%20Project.pdf 

I just wanted to share what I was able to create  using this library and thank anyone who was involved in the gampad libraries development. I hope the credit that I put in the report and on my github is sufficient and I would take any pointers on how I could do that better.

LeeNX, 2023-05-09:
That is awesome! @douggmill Are you going to add a ```Creative Commons copyright licenses```?  
  
This would be great as part of an awesome list of Projects, that people could get too.

douggmill, 2023-05-09:
Thank you! 
I added an Attribution 4.0 International creative commons license text file to the project and added the license badge to my readme file. I'm not sure about using the LEGO brand name in the project title though.

lemmingDev, 2024-02-24:
Glad the library came in handy

Here is a link to the project for anyone looking, as the link above no longer works due to a changed filename

https://github.com/douggmill/BLE-Arcade-Joystick-LEGO-Arduino

closed 2024-02-24 by lemmingDev