lemmingDev/ESP32-BLE-Gamepad #176: Alternative

lemmingDev, 2023-05-18:
FYI

The GP2040-CE firmware runs on the cheap Raspberry Pi Pico boards and is compatible with XInput etc and has less than 1ms latency wired. Anyone who needs low latency standard wired gamepad functionality might want to use that.
https://github.com/OpenStickCommunity/GP2040-CE

That firmware however only supports the standard gamepad layouts/number of buttons etc

They're looking to add Bluetooth support with the Pico W boards
https://github.com/OpenStickCommunity/GP2040-CE/issues/247

When that's done, I assume a lot of people who are after basic Bluetooth functionality will just use that

This library is still much more flexible though in that it supports configurable axes and buttons etc, though is no where near as compatible with non-Windows systems/consoles

LeeNX, 2023-05-24:
Seen it before, I think I have linked to in a few issues. Even got a few ideas from it. Looks cool, but BlueTooth is still a way off, I think. Worth keeping an eye on.

LeeNX, 2023-06-15:
Between the BLE examples - https://github.com/raspberrypi/pico-examples and something like the PR https://github.com/bluekitchen/btstack/pull/209. 
  
I think Pico BLE gamepad is getting closer.