lemmingDev/ESP32-BLE-Gamepad #177: Compiler stuck/not loading/ help for ps2-pc bridging?

Quarternary, 2023-05-21:
[image]

I am using an esp32 wroom-da module and have installed the latest version of the dependancy. However, the compiler keeps getting stuck with no progress even after 20 minutes. I am using the example code with no edits. I have tried different libraries for diffrent purposes but this problem does not exist. Please help me :D 

Also I found this code very complex as a complete noob to programming. My use case is to connect my old ps2 controller to my pc wirelessly. I have already set up and used the ps2x library for the esp 32 and am getting outputs in the serial monitor telling me all events like special button presses, analog values(0-255), presses for triangle, circle, square and x, all 4 trigger button presses, and dpad inputs. Could you provide susggestions / examples on how I can take these events and send them wirelessly?

Quarternary, 2023-05-23:
Has anyone ever experienced the compiler not loading before?

LeeNX, 2023-05-23:
Not had the Arduino IDE and tools hang on compiling. Had problem where wrong com port setup or incorrectly setup micro controller breaks a compile.  
  
Did you setup the ESP32 for Ardunio IDE - https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/installing.html

Have you been able to compile the Blink example with ESP32 board selected?

lemmingDev, 2023-05-23:
I've never used Arduino 2.x

Only Arduino 1.x and PlatformIO

Quarternary, 2023-05-23:
> Not had the Arduino IDE and tools hang on compiling. Had problem where wrong com port setup or incorrectly setup micro controller breaks a compile.
>
> Did you setup the ESP32 for Ardunio IDE - [...]

I have done it all properly for the esp 32, as mentioned before, I even got the esp32 to decode the output from a ps2 controller using another library. It works just fine with the latest versions of the ide too.

LeeNX, 2023-05-24:
Look at ```SingleButton.ino```, good starting point, at least for me ...

basically do something like 
```
  if(ps2x.Button(PSB_TRIANGLE))
    Serial.println("Triangle pressed");
    bleGamepad.press(BUTTON_1);
```
Also need to do the release part, for when the buttons are released.  
  
PS2 controller has analog buttons, only axes are analog at the moment, so the following would need to be switched to digital
```
if(ps2x.Button(PSB_PAD_UP)) {      
  Serial.print("Up held this hard: ");
  Serial.println(ps2x.Analog(PSAB_PAD_UP), DEC);
}
```

I think a look at ```Gamepad.ino``` has more buttons setup, I don't remember the maps of buttons to labels.  
  
Hope that helps, look forward to seeing your working project.

Quarternary, 2023-05-24:
so far great progress!! pulled an all-nighter and got all the buttons and stuff (including the analog buttons) set up perfectly. The only thing remaining is to set the axis. I even tested it out with x360ce. Thank you all for the help!!!

closed 2023-05-24 by Quarternary