lemmingDev/ESP32-BLE-Gamepad #179: Provide button map for android

Kashyapkoshti, 2023-06-04:
You have mentioned in features that, for android os device buttons configuration is different.

I am facing difficulty finding correct configuration for dpad and ABXY buttons.

Can someone help me with that.

If you can provide whole configuration list for each button or reference site for that. It will be very helpful.

LeeNX, 2023-06-14:
If you don't mind a little trial and error, maybe try [Game Controller Tester](https://play.google.com/store/apps/details?id=uk.co.powgames.gamecondiag&hl=en&gl=US) and draw up your findings.  
  
I also have been looking at things like https://www.w3.org/TR/gamepad/#fig-visual-representation-of-a-standard-gamepad-layout

LeeNX, 2023-06-14:
Hey @Kashyapkoshti , I did not really post any findings, I have only done very basic testing on Android myself. Did you try the TestAll sketch with Game Controller Tester?  
  
I think when I did this, I ran into a problem where a button kept escaping out of the test, but it mostly worked. Should track down the button and remove and test again.

Kashyapkoshti, 2023-06-14:
hi @LeeNX, i have used gamepad layout which you have provided. it is working in the windows machine but for Android, DPad is not working

closed 2023-12-25 by Kashyapkoshti