lemmingDev/ESP32-BLE-Gamepad #181: Controller doesn't work on games

Pyloupy, 2023-06-13:
Hello, 
I discover esp32 by this library because i want to create a simple controller like xbox. I started with just button A and B et 2 joystics. In a website tester it work perfectly well but in game like rocket league it doesn't work. In the game biped it recognize that there is a controller but when i press button or move joystics it doesn't work.

Any Idea of my problem ? (im sorry for my english it is not my native language.

Thanks :)

LeeNX, 2023-06-14:
What host Operating system you using? If it's Windows, does the Joystick tester pick up the BLE Gamepad? If you can run Steam, have you tested the Gamepad controller in the options menu?  
  
Have you tested on a mobile device, like Android? I like to use the [Game Controller Tester](https://play.google.com/store/apps/details?id=uk.co.powgames.gamecondiag&hl=en&gl=US)

Do you have any gamepad mappings tooling or something, that could be interfering?

Sean-Boyes, 2023-06-27:
I believe the game requires a xinput device/ a translation layer to work on games. Steam has a built in translation layer, just go to steam settings under controller and set it up.

lemmingDev, 2024-02-24:
As above

closed 2024-02-24 by lemmingDev