lemmingDev/ESP32-BLE-Gamepad #182: [Feature Request] Extended Battery support, like charging?

LeeNX, 2023-06-15:
Possible add GATT Battery Status (0x2A1A) and Battery Level State (0x2A1B) 

On the Bluetooth SIG page, https://www.bluetooth.com/specifications/specs/battery-service/ there is a release for [Battery Service 1.1 ](https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=554952)

Which includes some items that might be worth adding, if we want the host to know if the gamepad is charging or other battery health/service properties. It's more a UX thing, but I can see the gamepad been charged, while in use.

lemmingDev, 2025-02-13:
I am trying to add this at the moment
Hopefully will be done in a day or two

https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/266#issuecomment-2657877661

LeeNX, 2025-02-14:
This FR is mostly for UX, like I have seen on things like RetroPi, where the On-Screen-Display for battery level, which we do, but also if the device is been charged, with battery and lightning bolt displayed.  
  
If we get a bunch of other features with the same update, that is just awesome, but I am not sure that anybody is really using the battery health or battery serial numbers stuff.

lemmingDev, 2025-02-15:
Well, [releases from 7.3](https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.7.3) now have the ability to set 0x2A1A info, and have it available in the OS

0x2A1B seems to just be the level we already have, so I ignored that one
I tested adding other battery characteristics, which work, though I think this one is enough to give/set all the info we need

Check new [example](https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/SetBatteryPowerState/SetBatteryPowerState.ino) for info

Okay to close?

[image]

LeeNX, 2026-08-25:
This is feature complete, let's close issue for now.

closed 2026-08-25 by LeeNX