lemmingDev/ESP32-BLE-Gamepad #184: [Feature Request] Reference example for BLE blacklist?

LeeNX, 2023-06-26:
Maybe some ideas of how to possible temporary blacklist a host from connecting to gamepad.

Whiling looking to testing BLE HID/GATT/HOG info for #182 , I ran into a problem where another bluetooth host kept trying to pair with my test device. This was ignoring and I ended up giving up investigating, but later was thinking, if I could disable my gamepad from trying/allowing to pair with the bad/aggressive host, then I might not run into this problem.  
  
My idea, is that while the gamepad is in awaiting to pair mode, like asking to accept pairing, then if you hit some button, then for a set time, like 10min, gamepad would not allow or try to pair to the host.

Not sure if this is even possible, but an idea worth investigating and adding as an example to improve UX in general.

LeeNX, 2023-06-26:
Extending this idea a little further, but might be a pro user, expose a HID service, with the text info on what device the gamepad is paired with, something like the ```temperature broadcast``` service. Would be better than using USB serial feedback or even an eInk screen or possible a BLE serial interface. Maybe just make it a debug mode, that is off by default.

lemmingDev, 2025-01-29:
Closing this as it's been resolved v0.6.9
Check [ForcePairingMode.ino](https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/ForcePairingMode/ForcePairingMode.ino) example for details

closed 2025-01-29 by lemmingDev