lemmingDev/ESP32-BLE-Gamepad #185: Trying to get over 32 buttons to work.

Slythy, 2023-06-28:
I have modified the following line

`#define JOYSTICK_DEFAULT_BUTTON_COUNT       128
`
I have also found your previous code that had the line

```
Joystick_ Joystick(0x06, JOYSTICK_TYPE_GAMEPAD, 64, 0, 
  false, false, false, false, false, false, false, false, false, false, false);
```

I get the following error

```
`joystick:3:26: error: 'JOYSTICK_TYPE_GAMEPAD' was not declared in this scope
 Joystick_ Joystick(0x05, JOYSTICK_TYPE_GAMEPAD, 64, 0,
                          ^~~~~~~~~~~~~~~~~~~~~
Multiple libraries were found for "Joystick.h"
 Used: C:\Users\GrantGeorge\Documents\Arduino\libraries\Joystick
 Not used: C:\Users\GrantGeorge\Documents\Arduino\libraries\ArduinoJoystickLibrary-master
exit status 1
'JOYSTICK_TYPE_GAMEPAD' was not declared in this scope
```

Any Ideas? I am trying to read it on VKB button tester (MMJOY2) and Pointy's joystick tester only shows the 32 buttons. 

How can I get this to work? I am trying to get 14 buttons with 3 rotaries (28 buttons) and then 2 Hall sensors. 

Right now I am just trying 1 button below with the difinition changed from 32 to 64/128

```
#include <Joystick.h>

// Constant that maps the physical pin to the joystick button.
const int pinToButtonMap = 9;

void setup() {
  // Initialize Button Pins
  pinMode(pinToButtonMap, INPUT_PULLUP);

  // Initialize Joystick Library
  Joystick.begin();
}

// Last state of the button
int lastButtonState = 0;

void loop() {

  // Read pin values
  int currentButtonState = !digitalRead(pinToButtonMap);
  if (currentButtonState != lastButtonState)
  {
  Joystick.setButton(40, currentButtonState);
  lastButtonState = currentButtonState;
  }

  delay(50);
}
```

LeeNX, 2023-06-28:
I think you might be mixing your Arduino libraries?  
  
Start with the ```TestAll.ino``` example and change https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/fff571a54bf13b3503f43b9e59a96ef5fceec44b/examples/TestAll/TestAll.ino#L8 to the number of push button you want and compile from there.

LeeNX, 2023-06-28:
The above snippet is from [ArduinoJoystickLibrary](https://github.com/MHeironimus/ArduinoJoystickLibrary), not for ESP32-BLE-Gamepag. Best to open an issue in that project.

Best of luck

lemmingDev, 2024-02-24:
Thanks @LeeNX

closed 2024-02-24 by lemmingDev