lemmingDev/ESP32-BLE-Gamepad #189: Clarification of the library usage

dronecz, 2023-07-19:
Hello, 

I´m trying to build a BT controller for Retropie and I did find this excellent library. Sadly I´m a bit lost with the Arduino sketch and how to define and use Start and Select buttons. 

I did manage to set up the joystick and 6 buttons, but I do not know how to set up those 2 buttons.  I tried to find solutions in issues, in source code and on the Internet, but I did not find many examples of code to solve my problem. 

I have the following code:

```
/*
   This code programs several pins on an ESP32 as buttons and hats on a BLE gamepad

   It uses arrays to cut down on code

   Before using, adjust the numOfButtons, buttonPins, physicalButtons, numOfHats, and hatPins to suit your scenario

*/

#include <BleGamepad.h> // https://github.com/lemmingDev/ESP32-BLE-Gamepad

BleGamepad bleGamepad;

#define numOfButtons 6
#define numOfHats 1  // Maximum 4 hat switches supported

byte previousButtonStates[numOfButtons];
byte currentButtonStates[numOfButtons];
byte buttonPins[numOfButtons] = { 1, 2, 3, 4, 5, 6 };
byte physicalButtons[numOfButtons] = { 1, 2, 3, 4, 5, 6 };

byte previousHatStates[numOfHats * 4];
byte currentHatStates[numOfHats * 4];
byte hatPins[numOfHats * 4] = { 11, 12, 13, 14 }; // in order UP, LEFT, DOWN, RIGHT. 4 pins per hat switch (Eg. List 12 pins if there are 3 hat switches)

void setup()
{
  // Setup Buttons
  for (byte currentPinIndex = 0 ; currentPinIndex < numOfButtons ; currentPinIndex++)
  {
    pinMode(buttonPins[currentPinIndex], INPUT_PULLUP);
    previousButtonStates[currentPinIndex] = HIGH;
    currentButtonStates[currentPinIndex] =  HIGH;
  }

  // Setup Hat Switches
  for (byte currentPinIndex = 0 ; currentPinIndex < numOfHats * 4 ; currentPinIndex++)
  {
    pinMode(hatPins[currentPinIndex], INPUT_PULLUP);
    previousHatStates[currentPinIndex] = HIGH;
    currentHatStates[currentPinIndex] =  HIGH;
  }

  BleGamepadConfiguration bleGamepadConfig;
  bleGamepadConfig.setIncludeStart(true);
  bleGamepadConfig.setIncludeSelect(true);
  bleGamepadConfig.setAutoReport(false);
  bleGamepadConfig.setButtonCount(numOfButtons);
  bleGamepadConfig.setHatSwitchCount(numOfHats);
  bleGamepad.begin(&bleGamepadConfig);
  //bleGamepad.begin(numOfButtons, numOfHats);
  //  bleGamepad.setAutoReport(false);
}

void loop()
{
  if (bleGamepad.isConnected())
  {
    // Deal with buttons
    for (byte currentIndex = 0 ; currentIndex < numOfButtons ; currentIndex++)
    {
      currentButtonStates[currentIndex]  = digitalRead(buttonPins[currentIndex]);

      if (currentButtonStates[currentIndex] != previousButtonStates[currentIndex])
      {
        if (currentButtonStates[currentIndex] == LOW)
        {
          bleGamepad.press(physicalButtons[currentIndex]);
        }
        else
        {
          bleGamepad.release(physicalButtons[currentIndex]);
        }
      }
    }

    // Update hat switch pin states
    for (byte currentHatPinsIndex = 0 ; currentHatPinsIndex < numOfHats * 4 ; currentHatPinsIndex++)
    {
      currentHatStates[currentHatPinsIndex]  = digitalRead(hatPins[currentHatPinsIndex]);
    }

    // Update hats
    signed char hatValues[4] = { 0, 0, 0, 0 };

    for (byte currentHatIndex = 0 ; currentHatIndex < numOfHats ; currentHatIndex++)
    {
      signed char hatValueToSend = 0;

      for (byte currentHatPin = 0 ; currentHatPin < 4 ; currentHatPin++)
      {
        // Check for direction
        if (currentHatStates[currentHatPin + currentHatIndex * 4] == LOW)
        {
          hatValueToSend = currentHatPin * 2 + 1;

          // Account for the last diagonal
          if (currentHatPin == 0)
          {
            if (currentHatStates[currentHatIndex * 4 + 3] == LOW)
            {
              hatValueToSend = 8 ;
              break;
            }
          }

          // Account for the first 3 diagonals
          if (currentHatPin < 3)
          {
            if (currentHatStates[currentHatPin + currentHatIndex * 4 + 1] == LOW)
            {
              hatValueToSend += 1;
              break;
            }
          }
        }
      }

      hatValues[currentHatIndex] = hatValueToSend;
    }

    // Set hat values
    bleGamepad.setHats(hatValues[0], hatValues[1], hatValues[2], hatValues[3]);

    // Update previous states to current states and send report
    if (currentButtonStates != previousButtonStates || currentHatStates != previousHatStates)
    {
      for (byte currentIndex = 0 ; currentIndex < numOfButtons ; currentIndex++)
      {
        previousButtonStates[currentIndex] = currentButtonStates[currentIndex];
      }

      for (byte currentIndex = 0 ; currentIndex < numOfHats * 4 ; currentIndex++)
      {
        previousHatStates[currentIndex] = currentHatStates[currentIndex];
      }

      bleGamepad.sendReport();
    }

    delay(20);
  }
}
```

I can connect to ESP32, received all D-PAD axis, and A, B, C, X, Y, Z buttons. 

1) How do I set up a sketch so I can send just A, B, X, and Y buttons?
I started with just 4 buttons, but I was not getting X and Y buttons, which appear when I increased the number of buttons to 6, I would like to just define those A, B, X, and Y buttons. 

2) How do I set up the Start and Select buttons? 
I think I should do it with `if/else` statement in part of the code where buttons are handled. Is that the correct approach? 

Many thanks for your help.

LeeNX, 2023-07-19:
```ESP32-BLE-Gamepad``` does not config host OS button labels, this is up to the OS to interpret, label and assign. The library just sets up the predefined number of buttons - https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/fff571a54bf13b3503f43b9e59a96ef5fceec44b/BleGamepad.cpp#L177C13-L177C32 
 
I don't think without changing the HID setup, it would not be easy. In Retro Pie you should be able to config gamepad button assignment - https://retropie.org.uk/docs/Controller-Configuration/. 
Override the default buttons to the button you want.  
  
As for the two special button setup, the BLE part looks done, but you still need to config pins to monitor and then run ```bleGamepad.pressStart();``` or as the buttons press / release are needed for the state change of these buttons. An example of the BLE buttons - https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/fff571a54bf13b3503f43b9e59a96ef5fceec44b/examples/SpecialButtons/SpecialButtons.ino#L30 
Might just be easier to use Retro Pie button assignment feature with your current sketch.

Abdurashid200112, 2023-10-30:
Hi everyone, I can't use all the arrows.  i can use the x and y axes but not the other whites

lemmingDev, 2024-02-24:
> `ESP32-BLE-Gamepad` does not config host OS button labels, this is up to the OS to interpret, label and assign. The library just sets up the predefined number of buttons - [...]

Exactly

closed 2024-02-24 by lemmingDev