lemmingDev/ESP32-BLE-Gamepad #190: the ZL and ZR buttons created for nintendo swtich do not work but on my arcade board for blueretro they do not work

khtp, 2023-09-04:
BlueRetro firmware version
1.8.4

BlueRetro firmware specification
HW1

BlueRetro firmware variant
Universal

BlueRetro hardware type
External adapter dongle (1 port only)

Manufacturer
I made it myself

System used
Sony PlayStation 2 and gamecube

Bluetooth controller brand & name
playstation.bin and  gc-external.bin

What is problem? (only list ONE problem per report)
I built the arcade board compatible with the blueretro from scratch but I have a problem, all the buttons and joystick work without problems but when I add the ZL and ZR buttons and map them from blueretro.io as ZL KB:B and ZR KB:C and save changes in The ps2 console in games does not detect that button, that is, that mapping, the src laber: keyboard the other: psx-ps2 the payload that I used was frimware: 1.8.4 and payload: playstation.bin I want to know how it can be solved?

ZL KB:B and ZR KB:C has been: L2 Y R2 so as not to confuse the fact that on the board I already have the pins that would be RX2 Y D5 soldered, on nintendo swtich they do work but on ps2 and gamecube or any other console with blueretro it is not recognizing the L2 and R2 buttons that would be the RX2 and D5 pins and I just passed you where I am basing myself to make this board

What did you expect to happen?
I was hoping the buttons I added would work once I mapped it to blueretro.io, I'm just looking for them to work

[image]

[image]

[image]

lemmingDev, 2023-09-04:
Sorry - How exactly does this relate to the ESP32-BLE-Gamepad library?

khtp, 2023-09-05:
Isn't this related to the dark cloud project with blueretro? Excuse me, I thought you had created the mapping so that the arcade board recognizes it, what is the dark cloud blueretro programming, my problem is that pressing R Y ZL Y ZR buttons in blueretro mode does not work regardless of the current frimware, in nintendo switch console all the buttons mentioned work and the rest the problem happens when you use blueretro