lemmingDev/ESP32-BLE-Gamepad #194: Amazon Fire Stick, yes it works

phant0mias, 2023-10-16:
Hi,

with this library I managed to control a amazon fire stick. Thanks for the library.

Some features I still cannot use with bluetooth:

    controlling the volume up/down/mute
    on/off

After connecting the device in the fire stick settings as gamecontroller you can send the remote commands. Automatic reconnect after power down the TV and remote is working too.

You have to experiment with it and check the examples of this class.

I had an effort with the followings commands:

#include <BleGamepad.h>

BleGamepad bleGamepad

void setup()
{
.....
BleGamepadConfiguration bleGamepadConfig;
bleGamepadConfig.setAutoReport(true);
bleGamepadConfig.setControllerType(CONTROLLER_TYPE_GAMEPAD);
bleGamepadConfig.setIncludeStart(true);
bleGamepadConfig.setIncludeSelect(true);
bleGamepadConfig.setIncludeMenu(true);
bleGamepadConfig.setIncludeHome(true);
bleGamepadConfig.setIncludeBack(true);

bleGamepad.begin(&bleGamepadConfig);
....
}

as commands you can send:

// Back
bleGamepad.press(BUTTON_2);
delay( 20 );
bleGamepad.release(BUTTON_2);

// Down
bleGamepad.setHat1(HAT_DOWN);
delay( 20 );
bleGamepad.setHat1(HAT_CENTERED);

// Home
bleGamepad.press(BUTTON_12);
delay( 20 );
bleGamepad.release(BUTTON_12);

// Left
bleGamepad.setHat1(HAT_LEFT);
delay( 20 );
bleGamepad.setHat1(HAT_CENTERED);

// Menu
bleGamepad.press(BUTTON_11);
delay( 20 );
bleGamepad.release(BUTTON_11);

// OK
bleGamepad.press(BUTTON_1);
delay( 20 );
bleGamepad.release(BUTTON_1);

// Right
bleGamepad.setHat1(HAT_RIGHT);
delay( 20 );
bleGamepad.setHat1(HAT_CENTERED);

// Up
bleGamepad.setHat1(HAT_UP);
delay( 20 );
bleGamepad.setHat1(HAT_CENTERED);

LeeNX, 2023-10-17:
Have you tried? 
```
bleGamepadConfig.setIncludeVolumeInc(true);
bleGamepadConfig.setIncludeVolumeInc(true);
bleGamepadConfig.setIncludeVolumeMute(true);
```
Can be seen in the ```SpecialButtons``` sketch - https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/fff571a54bf13b3503f43b9e59a96ef5fceec44b/examples/SpecialButtons/SpecialButtons.ino#L39

phant0mias, 2023-10-17:
Hi,
yes I tried but it is not working

phant0mias, 2023-10-17:
But I am really happy that the other functions are working with the fire stick....

LeeNX, 2023-10-17:
Just for the record, do you know which version of FireOS you running?

phant0mias, 2023-10-17:
Yes, it is 6.7.0.0

Just for me as information. What should be the delay between press and release. I realized that without delay it was not working and with a delay of 10ms only sometimes.

So thanks for the library it was easy to handle the fire stick with it......