lemmingDev/ESP32-BLE-Gamepad #198: BSOD and WDF_VIOLATION error on Windows 11 after connected

praditautama, 2023-10-29:
Hello,
I am building BLE Gamepad with 0.9 inch OLED display on ESP32 DEVKITC V4 WROOM-32U, I use macOS for development. 

Everything works on macOS, I can't test all buttons using "Controller Lite" app from AppStore. But every time I connected to Windows 11 PC, I got BSOD with WDF_VIOLATION error. 

I searched for WDF_VIOLATION, It's related to bad driver mostly BT driver.  Is there any chance my USB dongle doesn't support for BLE?

My PC:
- Windows 11 with latest security updates
- TP-Link UB400. It's an BT 4.0 USB dongle. (https://www.tp-link.com/sg/home-networking/adapter/ub400/)

praditautama, 2023-10-29:
I tried to use `https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/111#issuecomment-1254260699` and i worked! 

is it related to NimBLE?

praditautama, 2023-10-30:
Hi @LeeNX I added my source code.

LeeNX, 2023-10-30:
Nothing jumps out as wrong or broken. 

Test the ```SingleButton``` sketch, to rule out Bluetooth hardware/OS/Drivers.  
  
If you not using HATs, could disable them.  
  
Interesting problem, keep us updated.

praditautama, 2023-11-01:
hi @LeeNX 

I bought another BT 5.x USB dongle and I don't get WDF_VIOLATION errors anymore. 

I think it's UB400 USD dongle issue. 

I will close this

closed 2023-11-01 by praditautama