lemmingDev/ESP32-BLE-Gamepad #199: Fight stick sketch request

thc1998, 2023-11-04:
I have no real technical ability to code I can modify some of the sketches to get some basic functionality but I was wondering if there is a sketch I can just define the pins for a 10 button 1 arcade stick setup with special function keys for start and select any guidance would be great I got the multi button example to work but I couldn't get the multi button plus hat to work at all

LeeNX, 2023-11-04:
I don't remember any flight stick sketches, but there are a few projects that might be worth looking at, like 

https://github.com/douggmill/BLE-Arcade-Joystick-LEGO-Arduino/blob/master/LEGO%20Arduino%20Arcade%20Joystick%20Home%20Project.pdf

or even

https://github.com/palsayantan/BLE-Game-Controller/blob/main/ESP32_BLE_Controller/ESP32_BLE_Controller.ino with a nice video https://www.youtube.com/watch?v=dODmsoAu0D4 

and possibly 

https://www.instructables.com/DIY-ESP32-Bluetooth-GamePad-for-Android-PlayStatio/
 
 If you find more suggestions, welcome to post them.

thc1998, 2023-11-09:
so after swapping bits of code and figuring out specifics i have found the triggers are swapped with the right thumb commands ill add an example of what i had to do to get triggers to work
if (digitalRead(LT) == LOW) {
      Serial.println(F("Left Trigger"));
      if (digitalRead(LT) == LOW) {
      bleGamepad.setRightThumb(32767, 0);      
      bleGamepad.sendReport();}
      else
      bleGamepad.setRightThumb(0, 0);  
      bleGamepad.sendReport();
    }
    if (digitalRead(RT) == LOW) {
      Serial.println(F("Right Trigger"));
      bleGamepad.setRightThumb(0, 32767);  
      bleGamepad.sendReport();}
      else
       bleGamepad.setRightThumb(0, 0);  
      bleGamepad.sendReport();
    }

lemmingDev, 2024-02-24:
Hi

I can confirm that the library is correctly assigning what Windows expects those axes to be

If you're on another operating system, they may interpret them differently

closed 2024-02-24 by lemmingDev