lemmingDev/ESP32-BLE-Gamepad #200: Special Buttons work everywhere but Netflix (so far)

Dark1886, 2023-11-15:
Remote Device: ESP32 Pico based
Device I'm controlling: Nvidia Shield TV Pro 

I'm testing this for a DIY remote and I'm noticing if i use the pressHome() and pressBack() functions they will work everywhere but inside of the Netflix app. I suspect this is Netflix taking control of a gamepad for it's own control and limiting the function of the home and back buttons. 

is there any way I may be able to get around this? My other option is likely to change to a keyboard based BLE library, but i'd prefer to not change my code. 

EDIT: It  appears like I may lose all bluetooth control using this library (arrow keys), but I thought that previously worked, so it could just be an issue with my code.

LeeNX, 2024-02-13:
Really up to the Host been connected to, if an application can disable or override the BLE device.  
  
Some ideas that I might try, would be trying to use another ```setControllerType```
```
bleGamepadConfig.setControllerType(CONTROLLER_TYPE_GAMEPAD); // CONTROLLER_TYPE_JOYSTICK, CONTROLLER_TYPE_GAMEPAD (DEFAULT), CONTROLLER_TYPE_MULTI_AXIS
```
Don't forget to unpair the BLE device between changes, else the same behavior might persist.  
  
If you tried the BLE keyboard or mouse to see if you get a similar problem?

lemmingDev, 2024-02-23:
@Dark1886
Any update on this?

Dark1886, 2024-02-23:
> @Dark1886
> 
> Any update on this?

I believe this is likely just an issue with how Netflix and other apps handle gamepads compared to other devices such as a keyboard. I believe it is intercepting those keystrokes and either canceling the home/back functionally to reduce accidentally clicking out, or repurposing it for in-app functions. 

As of now I am using the keyboard ble library and it is working as desired.

lemmingDev, 2024-02-24:
Thanks for getting back to me

Will mark it as closed then

closed 2024-02-24 by lemmingDev