lemmingDev/ESP32-BLE-Gamepad #204: Unable to use Rudder

iurycarlos, 2023-12-30:
Hello guys!

I took a look at the samples and used the Flight Simulator Test as starting point.
Removed all debug / log stuff and set pins and available buttons / axis.
However,  for some reason, i can't make RUDDER to appear at Windows Joystick Properties.

I only can see Throttle and Brake. Is there anything else we should set?

LeeNX, 2024-02-13:
I don't have Windows, so can't confirm the RUDDER not showing up. Have you tried to see what https://hardwaretester.com/gamepad on Chrome displays?
[image]

In the screenshot you can see Arduino serial console and the gamepad testing for the right side joystick, as an example.  
  
Labels and buttons are not standard and differ between platforms, thou I think @lemmingDev tests on Windows.

lemmingDev, 2024-02-24:
Windows doesn't show all available axes / buttons

If you enable throttle and rudder (no brake), you will see that the rudder displays fine

[image]

With throttle, rudder and brake enabled, all show in the gamepad tester above, and games would be able to access the axis

closed 2024-02-24 by lemmingDev