lemmingDev/ESP32-BLE-Gamepad #205: GamePad not visible

dimityrivanov, 2024-01-22:
When I compile the gamepad sketch on the ESP32 the device is not visible on any platform with bluetooth, any idea what might be the reason ?

dimityrivanov, 2024-01-28:
Ok I understood what the problem is its here xTaskCreate(this->taskServer, "server", 20000, (void *)this, 5, NULL) this (void *) this is not passed appropriately and if you execute the code without the xTask all works fine but it hangs here if you use the task  BleGamepad *BleGamepadInstance = (BleGamepad *)pvParameter;

@lemmingDev Any idea how to rewrite this to work ?

lemmingDev, 2024-01-29:
Hi

This library supports the standard ESP32 only, not the S3 or C3
If you work out how to get it working on other platforms such as the C3, feel free to put a pull request in

ebur644, 2024-01-30:
Hi,

I was having the same problem with the C3 platform, I changed the priority to be 1 instead of 5 and it worked. 

xTaskCreate(this->taskServer, "server", 20000, (void *)this,  1, NULL);

lemmingDev, 2024-01-31:
Thanks for the update - will get around to testing whether setting it to 1 works for regular ESP32, and if it does, will set it as the default.

lemmingDev, 2024-02-24:
It works, so I'll make it the default

Thanks!

closed 2024-02-24 by lemmingDev