lemmingDev/ESP32-BLE-Gamepad #206: Brand new

Cptnyar77, 2024-02-02:
Hi there. I've tried to find this in a search but haven't found an answer.

I'm trying to run the 4x4 keypad example. 

I'm using a DOIT board.
Buttons sourced from Amazon. 

I have not altered the code.

Wired up the button matrix on breadboard.

Testing via windows 10

Buttons 1,2,7 don't fire off at all.
Buttons 3 and 4 stay pressed once pressed.
All other buttons works just fine.

Any insight would be greatly appreciated.

Side note: the "dpad" sets to top left corner (is this normal? )

LeeNX, 2024-02-13:
Have you setup some debugging serial output, to confirm your keypad circuit is working correctly?

Another possibility is that some of your input pins have been used for another peripheral on your device.

lemmingDev, 2024-02-23:
Hi @Cptnyar77 

I suggest that you check the button presses independent of the ESP32-BLE-Gamepad library to remove that as being the problem from the equation. Once you're sure that all button presses are working correctly, then retry. Most likely a wiring or pin allocation issue as stated above.

As you haven't replied to the above suggestion, I will close for now. If you still need help, please reply and I will reopen.

closed 2024-02-23 by lemmingDev