lemmingDev/ESP32-BLE-Gamepad #207: Using BleGamepadConfiguration causes device to become un-responsive.

mrsnipey, 2024-02-03:
Hi.
I started with the Gamepad.ino example. Device worked and pressed buttons correctly.

However, if I use BleGamepadConfiguration to try and set any properties of the device, the code compiles and I can upload it but the device becomes completely un-responsive to any inputs and none of the settings are reflected (i.e. number of buttons / axes).

I have tried the 'TestAll.ino' sketch and get the same results.
Not sure if it's only me as no-one else seems to be getting this issue.

LeeNX, 2024-02-13:
Have you deleted/forgotten the device from your bluetooth manager on your host OS? The reason been that many (if not all) bluetooth managers cache the bluetooth device configuration.  
  
I have run into similar problem make changing to my gamepad configurations. Another idea is to change the VID/PID if you change device setup.

lemmingDev, 2024-02-23:
@mrsnipey  Let us know if deleting the device and pairing it again did not sort your issue.I will re-open the issue.

It needs to be done any time you change the number of buttons axis, and leave the same VID/PID, just as @LeeNX stated

closed 2024-02-23 by lemmingDev