lemmingDev/ESP32-BLE-Gamepad #208: BLE Gamepad that Works with ESP32S3

BrainBoardz, 2024-02-05:
Hi There: 

I'm looking for a wireless gamepad that will work with the ESP32S3. Looks like all of the Bluetooth Classic controllers are not compatible. Is it  possible to use the ESP32-BLE-Gamepad library to interface a BLE controller to ESP32S3 (not turn an ESP32S3 into a controller itself).   Would the 8Bitdo controllers work in this scenario? Any help would be appreciated.

Altarax, 2024-02-13:
Same here ! If you find a solution i'd be happy to know !

BrainBoardz, 2024-02-13:
Hi Altarax: 

I did have a conversation with the (very helpful) developer of BluePad32.  He suggested that the Terios T3 (the sell these on Aliexpress) should work (not a great unit but it does apparently work with the ESP32S3 using Bluepad32). The BD Address A0:5A:5D:28:DB:B1 btw. I've actually ordered a GameSir T3S (A0:5A:5D:28:DB:B1) as it is readily available from Amazon ($30  CAD). Might work!  Both controllers need to be in "home+X" mode according to the developer.  Those (possibly identical clones?) appear to be the only options other than discontinued controllers from Stadia.  Basically, it appears that at a minimum you would need a controller that supports Android devices. It must also support Bluetooth Low Energy as the S3 does not support classic bluetooth (a major PITA btw). I'll report back once I have had a chance to test the GameSir out.

LeeNX, 2024-02-13:
This discussion might be misplaced and not get much feedback, as this library is meant to be used to make gamepad type devices using an ESP32.  
  
```ESP32-BLE-Gamepad``` can't be used to connect to a BLE gamepad.

BrainBoardz, 2024-02-13:
Hi LeeNX

Agreed. Altarax, perhaps looks like it would be best to move this discussion over to the Bluepad32 project instead.  I'll update there when I have some details on the controller.

LeeNX, 2024-02-13:
When you want to make your own awesome gamepad, swig by and show us what you did.  
  
I have been looking at https://github.com/darthcloud/BlueRetro, which also looks pretty interesting.

emujicad, 2025-01-26:
Hello, did you test the controller? . I want to connect a gamepad ble to a rasbperry pi pico 2 W.

LeeNX, 2025-01-27:
@emujicad I think you might be in the wrong project, thou reading thou my comment, I am thinking you might be looking for confirmation that somebody else might have been able to use an ESP32-BLE-Gamepad project to connect to a Pico 2 W host?

This library is used to make gamepads type devices that use BLE to connect to a host, like a computer or console. Host can be another ESP32 device using [Bluepad32](https://github.com/ricardoquesada/bluepad32) too. I am working on something like this for battle bot type project.

I have not done this with a Pico device, but look at [Bluepad32 for Pico W](https://github.com/ricardoquesada/bluepad32/blob/main/examples/pico_w/README.md#bluepad32-for-pico-w)

If the Pico 2 W is BLE standard and works similar to ESP32, then I can confirm that an [ESP32-BLE-Gamepad](https://bluepad32.readthedocs.io/en/latest/supported_gamepads/#android
) project works or [Terios T3](https://bluepad32.readthedocs.io/en/latest/supported_gamepads/#misc-controllers) or an [XBox](https://bluepad32.readthedocs.io/en/latest/supported_gamepads/#xbox-wireless-model-1708-2-buttons
) Gamepad should work, as can be seen in the [Supported gamepads](https://bluepad32.readthedocs.io/en/latest/supported_gamepads/#supported-gamepads) page.

On another note, this issue is also closed, which means you might not get some feedback in a timely fashion. Hope you find what you need and can move forward with your project.