lemmingDev/ESP32-BLE-Gamepad #209: Macbook only sees Button 1 press/release

ryanmfransen, 2024-02-12:
Hi. Great library.  Wanted to check in with you on something - I am having an issue where my code isn't seeing anything but button 1 presses.

I loaded the following code I found on another issue to test it, and changed the pow(2,i) to just i, which I assume is from an old version?

Using https://hardwaretester.com/gamepad

It only shows b0 toggling, nothing else.

Any suggestions?

I am using the following lib deps:
lib_deps = 
	https://github.com/h2zero/NimBLE-Arduino
	https://github.com/lemmingDev/ESP32-BLE-Gamepad

Thanks!

Code used:
#include <BleGamepad.h> 

BleGamepad bleGamepad;

void setup() 
{
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleGamepad.begin();
}

void loop() 
{
  if(bleGamepad.isConnected()) 
  {
    Serial.println("Press every even numbered button.");
    for(int i = 1; i<32 ; i+=2)
    {
      bleGamepad.press(i);
    }
    delay(500);

    Serial.println("Release every even numbered button.");
    for(int i = 1; i<32 ; i+=2)
    {
      bleGamepad.release(i);
    }
    delay(500);
  }
}

LeeNX, 2024-02-13:
I think you missing the ```bleGamepadConfig.setButtonCount(numOfButtons);``` before ```bleGamepad.begin();```, thou the default should be 16 buttons.  
  
Does your testing tool show the number of defined buttons? My go to sketch is the ```TestAll.ino```. Don't forget to unpair your BLE device and pair again, so that the BLE attribute caching does not cause any problem, like the BLE device profile not updating.

ryanmfransen, 2024-02-13:
Ah, the subtlety I overlooked was button registration occurring in pairing and not connection.  I had increased the button count after first pair.

It works great now. Thank you! 

Again, great library, I am using this in a few projects I will link out when done. Really appreciate your help and all your work in the library.

LeeNX, 2024-02-13:
You welcome. Drop some links to your projects, always interested.  
  
Yes, thanks @lemmingDev for an awesome project.  
 
@ryanmfransen we can close the issue?

ryanmfransen, 2024-02-13:
Solution:  Ensure to re-pair device if config changes.