lemmingDev/ESP32-BLE-Gamepad #210: Fightstick Steam

ArmandoCZ06, 2024-02-24:
im traying to make a Fightstick using Gamepad Example. but doesnt work properly.

it just need
DPad
11 action buttons.

To be used on steam

Seems to be a problem related to thumbs or analog sticks because inputs went crazy.

help!

lemmingDev, 2024-02-24:
The file you loaded is working perfectly, showing all inputs moving

If you want to make a fightstick, you should probably start with another sketch such as MultipleButtonsAndHats or MultipleButtonsDebounce

Please send me the pins numbers you want to use for:
up, down, left, right
and buttons 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11

ArmandoCZ06, 2024-02-24:
Sorry if im not clear enough. English isn't my native language. i want to
use this config or at least, pretty similar because is the one actually is
 on my controller.

The current version my controller has,  disconect every 20 minutes or less
and its really anoying. I dont know why, but it happens, so thats why i
want to use your code, because seems to be better and more stable.
[image: image.png]
Reference:
https://yakaracolombia.github.io/esp32-online-tool/arcade2.html

thanks in advance

lemmingDev, 2024-02-24:
I only see 10 buttons -> start, select, a, b, x, y, l, r, zl, zr
What is button 11?

lemmingDev, 2024-02-24:
Try this

```
/*
 * This code programs a number of pins on an ESP32 as buttons and hats on a BLE gamepad
 *
 * It uses arrays to cut down on code
 *
 * Before using, adjust the numOfButtons, buttonPins, physicalButtons, numOfHats and hatPins to suit your scenario
 *
 */

#include <Arduino.h>
#include <BleGamepad.h> // https://github.com/lemmingDev/ESP32-BLE-Gamepad

BleGamepad bleGamepad;

#define numOfButtons 10
#define numOfHats 1 // Maximum 4 hat switches supported

byte previousButtonStates[numOfButtons];
byte currentButtonStates[numOfButtons];
byte buttonPins[numOfButtons] = {14, 12, 19, 17, 18, 23, 5, 16, 15, 21};
byte physicalButtons[numOfButtons] = {1, 2, 3, 4, 5, 6, 7, 8, 9, 10};

byte previousHatStates[numOfHats * 4];
byte currentHatStates[numOfHats * 4];
byte hatPins[numOfHats * 4] = {25, 26, 27, 22}; // in order UP, LEFT, DOWN, RIGHT. 4 pins per hat switch (Eg. List 12 pins if there are 3 hat switches)

void setup()
{
    // Setup Buttons
    for (byte currentPinIndex = 0; currentPinIndex < numOfButtons; currentPinIndex++)
    {
        pinMode(buttonPins[currentPinIndex], INPUT_PULLUP);
        previousButtonStates[currentPinIndex] = HIGH;
        currentButtonStates[currentPinIndex] = HIGH;
    }

    // Setup Hat Switches
    for (byte currentPinIndex = 0; currentPinIndex < numOfHats * 4; currentPinIndex++)
    {
        pinMode(hatPins[currentPinIndex], INPUT_PULLUP);
        previousHatStates[currentPinIndex] = HIGH;
        currentHatStates[currentPinIndex] = HIGH;
    }

    BleGamepadConfiguration bleGamepadConfig;
    bleGamepadConfig.setAutoReport(false);
    bleGamepadConfig.setButtonCount(numOfButtons);
    bleGamepadConfig.setHatSwitchCount(numOfHats);
    bleGamepad.begin(&bleGamepadConfig);

    // changing bleGamepadConfig after the begin function has no effect, unless you call the begin function again
}

void loop()
{
    if (bleGamepad.isConnected())
    {
        // Deal with buttons
        for (byte currentIndex = 0; currentIndex < numOfButtons; currentIndex++)
        {
            currentButtonStates[currentIndex] = digitalRead(buttonPins[currentIndex]);

            if (currentButtonStates[currentIndex] != previousButtonStates[currentIndex])
            {
                if (currentButtonStates[currentIndex] == LOW)
                {
                    bleGamepad.press(physicalButtons[currentIndex]);
                }
                else
                {
                    bleGamepad.release(physicalButtons[currentIndex]);
                }
            }
        }

        // Update hat switch pin states
        for (byte currentHatPinsIndex = 0; currentHatPinsIndex < numOfHats * 4; currentHatPinsIndex++)
        {
            currentHatStates[currentHatPinsIndex] = digitalRead(hatPins[currentHatPinsIndex]);
        }

        // Update hats
        signed char hatValues[4] = {0, 0, 0, 0};

        for (byte currentHatIndex = 0; currentHatIndex < numOfHats; currentHatIndex++)
        {
            signed char hatValueToSend = 0;

            for (byte currentHatPin = 0; currentHatPin < 4; currentHatPin++)
            {
                // Check for direction
                if (currentHatStates[currentHatPin + currentHatIndex * 4] == LOW)
                {
                    hatValueToSend = currentHatPin * 2 + 1;

                    // Account for last diagonal
                    if (currentHatPin == 0)
                    {
                        if (currentHatStates[currentHatIndex * 4 + 3] == LOW)
                        {
                            hatValueToSend = 8;
                            break;
                        }
                    }

                    // Account for first 3 diagonals
                    if (currentHatPin < 3)
                    {
                        if (currentHatStates[currentHatPin + currentHatIndex * 4 + 1] == LOW)
                        {
                            hatValueToSend += 1;
                            break;
                        }
                    }
                }
            }

            hatValues[currentHatIndex] = hatValueToSend;
        }

        // Set hat values
        bleGamepad.setHats(hatValues[0], hatValues[1], hatValues[2], hatValues[3]);

        // Update previous states to current states and send report
        if (currentButtonStates != previousButtonStates || currentHatStates != previousHatStates)
        {
            for (byte currentIndex = 0; currentIndex < numOfButtons; currentIndex++)
            {
                previousButtonStates[currentIndex] = currentButtonStates[currentIndex];
            }

            for (byte currentIndex = 0; currentIndex < numOfHats * 4; currentIndex++)
            {
                previousHatStates[currentIndex] = currentHatStates[currentIndex];
            }

            bleGamepad.sendReport();
        }

        delay(20);
    }
}
```

lemmingDev, 2024-02-26:
Hi

Did the code work for you?

ArmandoCZ06, 2024-02-26:
Hi
sorry.
im a newbie at coding and everthing else.
im trying to figure it out how does it work.

i just put the code, but i dont know whats some else to to. When it's
conected to PC, doesnt work.

mi esp32 is conected as described in diagram, to its fisical buttons

lemmingDev, 2024-02-26:
Did you un-pair the bluetooth device, and then re-pair?

It will not work until you do this

ArmandoCZ06, 2024-02-29:
Thank you so much

It works!

Thanks for your help. you are awesome-

ArmandoCZ06, 2024-03-08:
the last time, you help me alot whit the code, and i want to ask you
something about ESP32.
i use esp32, connected to a powerbank, to Bluetooth.
when i try to conect it via USB, it still working wireless, not wired as
intended.

do you know if there is a way to use it wireless when i plug in a powerbank
and wired when it's conected directly in the usb port?

lemmingDev, 2024-03-08:
Only some ESP32 can do USB HID and BT
The original ESP32 can't, and that's all I have
I don't have any ESP32-S3

Maybe it's possible, but I'm not sure sorry

ArmandoCZ06, 2024-03-16:
Hi.
Sorry for still asking for your help.
I'm a newbie and I really appreciate any help.

i just bought a ESP32 S3, and i want to load the code you had sent  to me.

can i use this pins in the code or is there a restriction?

Notes: i want to use it wired or Wireless (BT). Just like a XboX Controller.

[image: image.png]

[image: image.png]
thanks in advance.🙈

lemmingDev, 2024-03-16:
This library only works with Bluetooth LE wireless
For ESP32 S3, you should change the pins to the ones that are free on your board
[image]

closed 2024-03-28 by lemmingDev