lemmingDev/ESP32-BLE-Gamepad #212: Question: DPDA and AxeY

GahFlorencio, 2024-03-04:
Hello everyone, I'm working on a Guitar Hero controller and I'm using the left analog stick along with the D-pad, but only the analog stick is working. I've checked the inputs and they're receiving signals, but in the gamepad tester it's as if there's no D-pad. Did I make a mistake?

`if (bleGamepad.isConnected()) {
    // Deal with buttons
    for (byte currentIndex = 0; currentIndex < numOfButtons; currentIndex++) {
      currentButtonStates[currentIndex] = digitalRead(buttonPins[currentIndex]);

      if (currentButtonStates[currentIndex] != previousButtonStates[currentIndex]) {
        if (currentButtonStates[currentIndex] == LOW) {
          bleGamepad.press(physicalButtons[currentIndex]);
        } else {
          bleGamepad.release(physicalButtons[currentIndex]);
        }
      }
    }

    for (byte currentHatPinsIndex = 0; currentHatPinsIndex < numOfHats * 3; currentHatPinsIndex++) {
      currentHatStates[currentHatPinsIndex] = digitalRead(hatPins[currentHatPinsIndex]);
    }

    int hatValue = DPAD_CENTERED;

    if (currentHatStates[1] == LOW) {
      hatValue = DPAD_LEFT;
    }

    if (currentHatStates[0] == LOW) {
      hatValue = DPAD_UP;

      if (currentHatStates[1] == LOW) {
        hatValue = DPAD_UP_LEFT;
      }
    }

    if (currentHatStates[2] == LOW) {
      hatValue = DPAD_DOWN;

      if (currentHatStates[1] == LOW) {
        hatValue = DPAD_DOWN_LEFT;
      }
    }

    bleGamepad.setHat1(hatValue);

    int potValues[numberOfPotSamples];
    int potValue = 0;

    for (int i = 0; i < numberOfPotSamples; i++) {
      potValues[i] = analogRead(34);
      potValue += potValues[i];
      delay(delayBetweenSamples);
    }

    potValue = potValue / numberOfPotSamples;

    currentPot = map(potValue, 1200, 2200, 32737, 16368);

    bleGamepad.setY(currentPot);

    if (currentButtonStates != previousButtonStates || currentHatStates != previousHatStates || currentPot != previousPot) {
      for (byte currentIndex = 0; currentIndex < numOfButtons; currentIndex++) {
        previousButtonStates[currentIndex] = currentButtonStates[currentIndex];
      }

      for (byte currentIndex = 0; currentIndex < numOfHats * 3; currentIndex++) {
        previousHatStates[currentIndex] = currentHatStates[currentIndex];
      }

      previousPot = currentPot;

      bleGamepad.sendReport();
    }
  }`

lemmingDev, 2024-03-04:
Please post the full code

What OS are you using the controller on?

Have you tried un-pairing and re-pairing the device? (needs to be done every time the gamepad layout is changed)

The MultipleButtonsAndHats example definitely has working hats

GahFlorencio, 2024-03-04:
OS Ubuntu.

I tested it on https://hardwaretester.com/gamepad, https://www.onlinemictest.com/controller-tester/, and on the Android gamepad tester. All have the same result: axisY moves, but the D-pad doesn't

FULL CODE.

``` #include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <AsyncElegantOTA.h>
#include <Arduino.h>
#include <BleGamepad.h>  // https://github.com/lemmingDev/ESP32-BLE-Gamepad

BleGamepad bleGamepad("Guitar-Clone", "GCarmoDev", 100); 
BleGamepadConfiguration bleGamepadConfig;
#define numOfButtons 7
#define numOfHats 1  // Maximum 4 hat switches supported

byte previousButtonStates[numOfButtons];
byte currentButtonStates[numOfButtons];
byte buttonPins[numOfButtons] = { 13, 14, 16, 17, 18, 22, 23 };
byte physicalButtons[numOfButtons] = { 6, 2, 1, 3, 4, 10, 9 };

byte previousHatStates[numOfHats * 3];
byte currentHatStates[numOfHats * 3];
byte hatPins[numOfHats * 4] = { 19, 25, 21 };  // in order UP, LEFT, DOWN, RIGHT. 4 pins per hat switch (Eg. List 12 pins if there are 3 hat switches)

const int numberOfPotSamples = 5;      // Number of pot samples to take (to smooth the values)
const int delayBetweenSamples = 4;     // Delay in milliseconds between pot samples
const int delayBetweenHIDReports = 5;  // Additional delay in milliseconds between HID reports

int previousPot = 1200;
int currentPot = 1200;

const char* ssid = "sddi";
const char* password = "password";

AsyncWebServer server(80);

#define OTA_PIN 13

bool isOta = false;

void setup(void) {
  pinMode(OTA_PIN, INPUT_PULLUP);
  isOta = digitalRead(OTA_PIN) == LOW;

  if (isOta) {
      ota_setup();
  }else{
      ble_gamepad_setup();
  }
}

void loop(void) {
  if(!isOta){
    ble_gamepad_loop();
  }
}

void ota_setup(){
  Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    Serial.println("");

    // Wait for connection
    while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
    }
    Serial.println("");
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());

    AsyncElegantOTA.begin(&server);    
    server.begin();
    Serial.println("HTTP server started");
}

void ble_gamepad_setup() {
  for (byte currentPinIndex = 0; currentPinIndex < numOfButtons; currentPinIndex++) {
    pinMode(buttonPins[currentPinIndex], INPUT_PULLUP);
    previousButtonStates[currentPinIndex] = HIGH;
    currentButtonStates[currentPinIndex] = HIGH;
  }

  for (byte currentPinIndex = 0; currentPinIndex < numOfHats * 3; currentPinIndex++) {
    pinMode(hatPins[currentPinIndex], INPUT_PULLUP);
    previousHatStates[currentPinIndex] = HIGH;
    currentHatStates[currentPinIndex] = HIGH;
  }

  bleGamepadConfig.setAutoReport(false);
  bleGamepadConfig.setButtonCount(numOfButtons);
  bleGamepadConfig.setHatSwitchCount(numOfHats);
  bleGamepad.begin(&bleGamepadConfig);
}

void ble_gamepad_loop() {
  if (bleGamepad.isConnected()) {
    for (byte currentIndex = 0; currentIndex < numOfButtons; currentIndex++) {
      currentButtonStates[currentIndex] = digitalRead(buttonPins[currentIndex]);

      if (currentButtonStates[currentIndex] != previousButtonStates[currentIndex]) {
        if (currentButtonStates[currentIndex] == LOW) {
          bleGamepad.press(physicalButtons[currentIndex]);
        } else {
          bleGamepad.release(physicalButtons[currentIndex]);
        }
      }
    }

    for (byte currentHatPinsIndex = 0; currentHatPinsIndex < numOfHats * 3; currentHatPinsIndex++) {
      currentHatStates[currentHatPinsIndex] = digitalRead(hatPins[currentHatPinsIndex]);
    }

    int hatValue = DPAD_CENTERED;

    if (currentHatStates[1] == LOW) {
      hatValue = DPAD_LEFT;
    }

    if (currentHatStates[0] == LOW) {
      hatValue = DPAD_UP;

      if (currentHatStates[1] == LOW) {
        hatValue = DPAD_UP_LEFT;
      }
    }

    if (currentHatStates[2] == LOW) {
      hatValue = DPAD_DOWN;

      if (currentHatStates[1] == LOW) {
        hatValue = DPAD_DOWN_LEFT;
      }
    }

    bleGamepad.setHat1(hatValue);

    int potValues[numberOfPotSamples];
    int potValue = 0;

    for (int i = 0; i < numberOfPotSamples; i++) {
      potValues[i] = analogRead(34);
      potValue += potValues[i];
      delay(delayBetweenSamples);
    }

    potValue = potValue / numberOfPotSamples;

    currentPot = map(potValue, 1200, 2200, 32737, 16368);

    bleGamepad.setY(currentPot);

    if (currentButtonStates != previousButtonStates || currentHatStates != previousHatStates || currentPot != previousPot) {
      for (byte currentIndex = 0; currentIndex < numOfButtons; currentIndex++) {
        previousButtonStates[currentIndex] = currentButtonStates[currentIndex];
      }

      for (byte currentIndex = 0; currentIndex < numOfHats * 3; currentIndex++) {
        previousHatStates[currentIndex] = currentHatStates[currentIndex];
      }

      previousPot = currentPot;

      bleGamepad.sendReport();
    }
  }
}

```

lemmingDev, 2024-03-04:
I removed the OTA stuff as I don't have all of those libraries, and it works fine on Windows (pic of me pressing the left dpad below)
The first gamepad tester you linked showed a change as axis 9, however the second one didn't respond at all
If Ubuntu doesn't detect the dpad, maybe just add more buttons and map them in the game

[image]

Looking at your code, you changed all of your dpad arrays to remove the ability to detect right d-pad presses. Intentional?

Code I slightly modifies to remove OTA stuff below, though I think it's Ubuntu, not the code

```
#include <Arduino.h>
#include <BleGamepad.h>  // https://github.com/lemmingDev/ESP32-BLE-Gamepad

BleGamepad bleGamepad("Guitar-Clone", "GCarmoDev", 100); 
BleGamepadConfiguration bleGamepadConfig;
#define numOfButtons 7
#define numOfHats 1  // Maximum 4 hat switches supported

byte previousButtonStates[numOfButtons];
byte currentButtonStates[numOfButtons];
byte buttonPins[numOfButtons] = { 13, 14, 16, 17, 18, 22, 23 };
byte physicalButtons[numOfButtons] = { 6, 2, 1, 3, 4, 10, 9 };

byte previousHatStates[numOfHats * 3];
byte currentHatStates[numOfHats * 3];
byte hatPins[numOfHats * 4] = { 19, 25, 21 };  // in order UP, LEFT, DOWN, RIGHT. 4 pins per hat switch (Eg. List 12 pins if there are 3 hat switches)

const int numberOfPotSamples = 5;      // Number of pot samples to take (to smooth the values)
const int delayBetweenSamples = 4;     // Delay in milliseconds between pot samples
const int delayBetweenHIDReports = 5;  // Additional delay in milliseconds between HID reports

int previousPot = 1200;
int currentPot = 1200;

void setup() {
  for (byte currentPinIndex = 0; currentPinIndex < numOfButtons; currentPinIndex++) {
    pinMode(buttonPins[currentPinIndex], INPUT_PULLUP);
    previousButtonStates[currentPinIndex] = HIGH;
    currentButtonStates[currentPinIndex] = HIGH;
  }

  for (byte currentPinIndex = 0; currentPinIndex < numOfHats * 3; currentPinIndex++) {
    pinMode(hatPins[currentPinIndex], INPUT_PULLUP);
    previousHatStates[currentPinIndex] = HIGH;
    currentHatStates[currentPinIndex] = HIGH;
  }

  bleGamepadConfig.setAutoReport(false);
  bleGamepadConfig.setButtonCount(numOfButtons);
  bleGamepadConfig.setHatSwitchCount(numOfHats);
  bleGamepad.begin(&bleGamepadConfig);
}

void loop() {
  if (bleGamepad.isConnected()) {
    for (byte currentIndex = 0; currentIndex < numOfButtons; currentIndex++) {
      currentButtonStates[currentIndex] = digitalRead(buttonPins[currentIndex]);

      if (currentButtonStates[currentIndex] != previousButtonStates[currentIndex]) {
        if (currentButtonStates[currentIndex] == LOW) {
          bleGamepad.press(physicalButtons[currentIndex]);
        } else {
          bleGamepad.release(physicalButtons[currentIndex]);
        }
      }
    }

    for (byte currentHatPinsIndex = 0; currentHatPinsIndex < numOfHats * 3; currentHatPinsIndex++) {
      currentHatStates[currentHatPinsIndex] = digitalRead(hatPins[currentHatPinsIndex]);
    }

    int hatValue = DPAD_CENTERED;

    if (currentHatStates[1] == LOW) {
      hatValue = DPAD_LEFT;
    }

    if (currentHatStates[0] == LOW) {
      hatValue = DPAD_UP;

      if (currentHatStates[1] == LOW) {
        hatValue = DPAD_UP_LEFT;
      }
    }

    if (currentHatStates[2] == LOW) {
      hatValue = DPAD_DOWN;

      if (currentHatStates[1] == LOW) {
        hatValue = DPAD_DOWN_LEFT;
      }
    }

    bleGamepad.setHat1(hatValue);

    int potValues[numberOfPotSamples];
    int potValue = 0;

    for (int i = 0; i < numberOfPotSamples; i++) {
      potValues[i] = analogRead(34);
      potValue += potValues[i];
      delay(delayBetweenSamples);
    }

    potValue = potValue / numberOfPotSamples;

    currentPot = map(potValue, 1200, 2200, 32737, 16368);

    bleGamepad.setY(currentPot);

    if (currentButtonStates != previousButtonStates || currentHatStates != previousHatStates || currentPot != previousPot) {
      for (byte currentIndex = 0; currentIndex < numOfButtons; currentIndex++) {
        previousButtonStates[currentIndex] = currentButtonStates[currentIndex];
      }

      for (byte currentIndex = 0; currentIndex < numOfHats * 3; currentIndex++) {
        previousHatStates[currentIndex] = currentHatStates[currentIndex];
      }

      previousPot = currentPot;

      bleGamepad.sendReport();
    }
  }
}
```

lemmingDev, 2024-03-04:
To clarify, the USB HID is set for a dpad. It's up to the OS to interpret it as it sees fit.

The HID uses logical min-max of 1-8 and physical min-max of 0-315 as rotational position

If another combination is more compatible with Linux, though still compatible with Windows/Android, let me know and I'll test it out, and make it the default

GahFlorencio, 2024-03-04:
Yes, I intentionally removed it because the Guitar Hero controller only contains dpad-up and dpad-down. For the PS2 to recognize it as a Guitar Hero controller, dpad-left needs to be pressed. I'm adapting a PS2 guitar to work both on the PS2 via Blue Retro and on PC and Android. 
 I will test it on Windows. What surprises me is that it didn't work on Android.
Thanks

GahFlorencio, 2024-03-06:
Tested it on Windows using both Windows and jstest-gtk (a Linux app), and they worked perfectly. I also tested it on the Linux app Guitar-Clone, and it was recognized within it. I believe the issue lies with the gamepad testing websites. Thank you very much; it worked very well. I suggest using jstest-gtk for Linux OS when testing BLEGamepad functionalities.

[image]

closed 2024-03-06 by GahFlorencio