lemmingDev/ESP32-BLE-Gamepad #213: Trying to make 32 button 3 axis game controller

dwizle81, 2024-03-13:
Library 5.4
IDE 2.3.2
board  ESP32 Wroom breakout board 
using the flight controller example straight from the example
code will never compile  keeps giving a ping time-out error or 
/Users/dwayne/Documents/Arduino/libraries/ESP32-BLE-Gamepad/BleGamepadConfiguration.cpp: In constructor 'BleGamepadConfiguration::BleGamepadConfiguration()':
/Users/dwayne/Documents/Arduino/libraries/ESP32-BLE-Gamepad/BleGamepadConfiguration.cpp:22:79: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
                                                      _hardwareRevision("1.0.0")
                                                                               ^
/Users/dwayne/Documents/Arduino/libraries/ESP32-BLE-Gamepad/BleGamepadConfiguration.cpp:22:79: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/Users/dwayne/Documents/Arduino/libraries/ESP32-BLE-Gamepad/BleGamepadConfiguration.cpp:22:79: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/Users/dwayne/Documents/Arduino/libraries/ESP32-BLE-Gamepad/BleGamepadConfiguration.cpp:22:79: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
/Users/dwayne/Documents/Arduino/libraries/ESP32-BLE-Gamepad/BleGamepadConfiguration.cpp:22:79: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]/Users/dwayne/Desktop/sketch_mar13a/sketch_mar13a.ino:2:11: error: expected constructor, destructor, or type conversion before 'controller'
  * Flight controller test
           ^~~~~~~~~~

lemmingDev, 2024-03-14:
Compiles fine for me in Arduino 1.8.19 and also the Arduino web editor

dwizle81, 2024-03-16:
> Compiles fine for me in Arduino 1.8.19 and also the Arduino web editor 
I got my code to compile uninstalled the IDE, deleted all the leftover folders did a new install
now my issue is it compiles it connects once but will not reconnect if you remove power.
what am I doing wrong??

#include <Arduino.h>
#include <BleGamepad.h>
#include <Keypad.h>
//#include <ESP32RotaryEncoder.h>

#define NUM_OF_BUTTONS 36
#define NUM_OF_HAT_SWITCHES 0

BleGamepad bleGamepad("BLE Flight Controller", "lemmingDev", 100);
BleGamepadConfiguration bleGamepadConfig;

const int potRY = 36; // Analog pin for RY axis (potentiometer)
const int potRX = 39; // Analog pin for RX axis (potentiometer)
const int potRZ = 34; // Analog pin for RZ axis (potentiometer)

const int numberOfPotSamples = 5;      // Number of pot samples to take (to smooth the values)
const int delayBetweenSamples = 4;     // Delay in milliseconds between pot samples
const int delayBetweenHIDReports = 5;  // Additional delay in milliseconds between HID reports
const int debounceDelay = 10;          // Delay in milliseconds between button press

const byte ROWS = 6; // Six rows
const byte COLS = 5; // Five columns
char keys[ROWS][COLS] = {
    {'1', '2', '3', '4', '5'},
    {'6', '7', '8', '9', '0'},
    {'A', 'B', 'C', 'D', 'E'},
    {'F', 'G', 'H', 'I', 'J'},
    {'K', 'L', 'M', 'N', 'O'},
    {'P', 'Q', 'R', 'S', 'T'}};
byte rowPins[ROWS] = {25, 26, 27, 32, 33, 18};    // Connect to the row pinouts of the keypad
byte colPins[COLS] = {2, 15, 12, 4, 5};    // Connect to the column pinouts of the keypad

Keypad customKeypad = Keypad(makeKeymap(keys), rowPins, colPins, ROWS, COLS);

void setup() {
    Serial.begin(115200);
    Serial.println("Starting BLE work!");

    // Set BLE characteristic options
    bleGamepadConfig.setAutoReport(true);
    bleGamepadConfig.setControllerType(CONTROLLER_TYPE_MULTI_AXIS); // Multi-axis joystick
    bleGamepadConfig.setButtonCount(NUM_OF_BUTTONS);
    bleGamepadConfig.setHatSwitchCount(NUM_OF_HAT_SWITCHES);
    bleGamepadConfig.setVid(0xe502);
    bleGamepadConfig.setPid(0xabcd);
    bleGamepadConfig.setAxesMin(0x0000); // Axes minimum value
    bleGamepadConfig.setAxesMax(0x7FFF); // Axes maximum value
    
    bleGamepadConfig.setModelNumber("1.0");
    bleGamepadConfig.setSoftwareRevision("Software Rev 1");
    bleGamepadConfig.setSerialNumber("9876543210");
    bleGamepadConfig.setFirmwareRevision("2.0");
    bleGamepadConfig.setHardwareRevision("1.7");
    
    bleGamepadConfig.setWhichAxes(true, true, true, true, true, true, false, false); // Enable RX, RY, RZ axes

    // Set custom MAC address
    uint8_t newMACAddress[] = {0xAA, 0xBB, 0xCC, 0xDD, 0xEE, 0xFD};
    esp_base_mac_addr_set(&newMACAddress[0]);

    // Begin BLE gamepad with configured settings
    bleGamepad.begin(&bleGamepadConfig);
}

void loop() {
    if (bleGamepad.isConnected()) {
        Serial.println("Connected to BLE gamepad!");

        char key = customKeypad.getKey();

        if (key != NO_KEY) {
            int buttonNum = (key - 'A') + 1; // Convert the keypad key to button number

            bleGamepad.press(buttonNum);
            bleGamepad.sendReport();
            delay(100);
            bleGamepad.release(buttonNum);
            bleGamepad.sendReport();
            delay(25);
        }

        // Check if any potentiometers are rotated
        int potValueRY = analogRead(potRY);
        int potValueRX = analogRead(potRX);
        int potValueRZ = analogRead(potRZ);

        bleGamepad.setRY(potValueRY);
        bleGamepad.setRX(potValueRX);
        bleGamepad.setRZ(potValueRZ);

        bleGamepad.sendReport();

        delay(1000); // Add a delay to control the update rate
    } else {
        Serial.println("BLE gamepad disconnected!");
        delay(1000); // Add a delay before trying to reconnect
    }
}

lemmingDev, 2024-03-16:
Please post your code between triple backtags ` so it shows correctly...

Your code compiles and I can pair it, but doesn't show up as a controller for me

I tried removing all of the custom vid/pid/serial etc stuff and it's working

Also, you set bleGamepadConfig.setAutoReport(true);, but then you manually call bleGamepad.sendReport();
I set auto report to false below
I changed the delay to 20ms, which polls approx 50 times per sec

Try the code below and see if it works
```
#include <Arduino.h>
#include <BleGamepad.h>
#include <Keypad.h>
//#include <ESP32RotaryEncoder.h>

#define NUM_OF_BUTTONS 36
#define NUM_OF_HAT_SWITCHES 0

BleGamepad bleGamepad("BLE Flight Controller", "lemmingDev", 100);
BleGamepadConfiguration bleGamepadConfig;

const int potRY = 36; // Analog pin for RY axis (potentiometer)
const int potRX = 39; // Analog pin for RX axis (potentiometer)
const int potRZ = 34; // Analog pin for RZ axis (potentiometer)

const int numberOfPotSamples = 5; // Number of pot samples to take (to smooth the values)
const int delayBetweenSamples = 4; // Delay in milliseconds between pot samples
const int delayBetweenHIDReports = 5; // Additional delay in milliseconds between HID reports
const int debounceDelay = 10; // Delay in milliseconds between button press

const byte ROWS = 6; // Six rows
const byte COLS = 5; // Five columns
char keys[ROWS][COLS] = {
{'1', '2', '3', '4', '5'},
{'6', '7', '8', '9', '0'},
{'A', 'B', 'C', 'D', 'E'},
{'F', 'G', 'H', 'I', 'J'},
{'K', 'L', 'M', 'N', 'O'},
{'P', 'Q', 'R', 'S', 'T'}};
byte rowPins[ROWS] = {25, 26, 27, 32, 33, 18}; // Connect to the row pinouts of the keypad
byte colPins[COLS] = {2, 15, 12, 4, 5}; // Connect to the column pinouts of the keypad

Keypad customKeypad = Keypad(makeKeymap(keys), rowPins, colPins, ROWS, COLS);

void setup() {
Serial.begin(115200);
Serial.println("Starting BLE work!");

// Set BLE characteristic options
bleGamepadConfig.setControllerType(CONTROLLER_TYPE_MULTI_AXIS); // Multi-axis joystick
bleGamepadConfig.setButtonCount(NUM_OF_BUTTONS);
bleGamepadConfig.setHatSwitchCount(NUM_OF_HAT_SWITCHES);
bleGamepadConfig.setAxesMin(0x0000); // Axes minimum value
bleGamepadConfig.setAxesMax(0x7FFF); // Axes maximum value
bleGamepadConfig.setWhichAxes(true, true, true, false, false, false, false, false); // Enable RX, RY, RZ axes

// Begin BLE gamepad with configured settings
bleGamepad.begin(&bleGamepadConfig);
}

void loop() {
  if (bleGamepad.isConnected()) {
  Serial.println("Connected to BLE gamepad!");

    char key = customKeypad.getKey();

    if (key != NO_KEY) {
        int buttonNum = (key - 'A') + 1; // Convert the keypad key to button number

        bleGamepad.press(buttonNum);
        bleGamepad.sendReport();
        delay(100);
        bleGamepad.release(buttonNum);
        bleGamepad.sendReport();
        delay(25);
    }

    // Check if any potentiometers are rotated
    int potValueRY = analogRead(potRY);
    int potValueRX = analogRead(potRX);
    int potValueRZ = analogRead(potRZ);

    bleGamepad.setRY(potValueRY);
    bleGamepad.setRX(potValueRX);
    bleGamepad.setRZ(potValueRZ);

    bleGamepad.sendReport();

    delay(20); // Add a delay to control the update rate
  } else {
    Serial.println("BLE gamepad disconnected!");
    delay(1000); // Add a delay before trying to reconnect
  }
}
```

lemmingDev, 2024-03-16:
Also, your analogs won't move much as analog read will return a value from 0 to 4095, though you've set the setAxesMax to 0x7FFF which is 32767

You should change:

bleGamepadConfig.setAxesMax(0x7FFF); // Axes maximum value

to 

bleGamepadConfig.setAxesMax(0x0FFF); // Axes maximum value****

lemmingDev, 2024-03-28:
So, I'll close this soon if I don't hear back

closed 2024-11-23 by lemmingDev