lemmingDev/ESP32-BLE-Gamepad #214: Trying to make a simple wired 4 button gamepad on a ESP32-WROVER-E board (Deneyap Kart)

Grandmaquil, 2024-03-15:
i am new to arduino stuff so help would be appriciated

lemmingDev, 2024-03-15:
This library is for making **wireless** Bluetooth gamepads...
Change the pin numbers to suit your board
Remember to unpair the Bluetooth before uploading new code 

```
#include <Arduino.h>
#include <BleGamepad.h> // https://github.com/lemmingDev/ESP32-BLE-Gamepad

BleGamepad bleGamepad("BLE Fightstick", "lemmingDev", 100);

#define numOfButtons 4
#define numOfHats 1

byte previousButtonStates[numOfButtons];
byte currentButtonStates[numOfButtons];
byte buttonPins[numOfButtons] = {0, 1, 8, 9}; // Button pins 1, 2, 3, 4
byte physicalButtons[numOfButtons] = {1, 2, 3, 4};

byte previousHatStates[numOfHats * 4];
byte currentHatStates[numOfHats * 4];
byte hatPins[numOfHats * 4] = {12, 13, 14, 15}; // DPAD direction pins UP, LEFT, DOWN, RIGHT

void setup()
{
    // Setup Buttons
    for (byte currentPinIndex = 0; currentPinIndex < numOfButtons; currentPinIndex++)
    {
        pinMode(buttonPins[currentPinIndex], INPUT_PULLUP);
        previousButtonStates[currentPinIndex] = HIGH;
        currentButtonStates[currentPinIndex] = HIGH;
    }

    // Setup Hat Switches
    for (byte currentPinIndex = 0; currentPinIndex < numOfHats * 4; currentPinIndex++)
    {
        pinMode(hatPins[currentPinIndex], INPUT_PULLUP);
        previousHatStates[currentPinIndex] = HIGH;
        currentHatStates[currentPinIndex] = HIGH;
    }

    BleGamepadConfiguration bleGamepadConfig;
    bleGamepadConfig.setAutoReport(false);
    bleGamepadConfig.setWhichAxes(0, 0, 0, 0, 0, 0, 0, 0);  // Disable all axes
    bleGamepadConfig.setButtonCount(numOfButtons);
    bleGamepadConfig.setHatSwitchCount(numOfHats);
    bleGamepad.begin(&bleGamepadConfig);

    // changing bleGamepadConfig after the begin function has no effect, unless you call the begin function again
}

void loop()
{
    if (bleGamepad.isConnected())
    {
        // Deal with buttons
        for (byte currentIndex = 0; currentIndex < numOfButtons; currentIndex++)
        {
            currentButtonStates[currentIndex] = digitalRead(buttonPins[currentIndex]);

            if (currentButtonStates[currentIndex] != previousButtonStates[currentIndex])
            {
                if (currentButtonStates[currentIndex] == LOW)
                {
                    bleGamepad.press(physicalButtons[currentIndex]);
                }
                else
                {
                    bleGamepad.release(physicalButtons[currentIndex]);
                }
            }
        }

        // Update hat switch pin states
        for (byte currentHatPinsIndex = 0; currentHatPinsIndex < numOfHats * 4; currentHatPinsIndex++)
        {
            currentHatStates[currentHatPinsIndex] = digitalRead(hatPins[currentHatPinsIndex]);
        }

        // Update hats
        signed char hatValues[4] = {0, 0, 0, 0};

        for (byte currentHatIndex = 0; currentHatIndex < numOfHats; currentHatIndex++)
        {
            signed char hatValueToSend = 0;

            for (byte currentHatPin = 0; currentHatPin < 4; currentHatPin++)
            {
                // Check for direction
                if (currentHatStates[currentHatPin + currentHatIndex * 4] == LOW)
                {
                    hatValueToSend = currentHatPin * 2 + 1;

                    // Account for last diagonal
                    if (currentHatPin == 0)
                    {
                        if (currentHatStates[currentHatIndex * 4 + 3] == LOW)
                        {
                            hatValueToSend = 8;
                            break;
                        }
                    }

                    // Account for first 3 diagonals
                    if (currentHatPin < 3)
                    {
                        if (currentHatStates[currentHatPin + currentHatIndex * 4 + 1] == LOW)
                        {
                            hatValueToSend += 1;
                            break;
                        }
                    }
                }
            }

            hatValues[currentHatIndex] = hatValueToSend;
        }

        // Set hat values
        bleGamepad.setHats(hatValues[0], hatValues[1], hatValues[2], hatValues[3]);

        // Update previous states to current states and send report
        if (currentButtonStates != previousButtonStates || currentHatStates != previousHatStates)
        {
            for (byte currentIndex = 0; currentIndex < numOfButtons; currentIndex++)
            {
                previousButtonStates[currentIndex] = currentButtonStates[currentIndex];
            }

            for (byte currentIndex = 0; currentIndex < numOfHats * 4; currentIndex++)
            {
                previousHatStates[currentIndex] = currentHatStates[currentIndex];
            }

            bleGamepad.sendReport();
        }

        delay(20);
    }
}
```

lemmingDev, 2024-03-18:
Hi

Please provide an update

Grandmaquil, 2024-03-19:
sorry. i was looking for something wired, thanks for your help

LeeNX, 2024-03-19:
@Grandmaquil - might be worth looking at https://gp2040-ce.info/ if you not locked into using an ESP32

Grandmaquil, 2024-03-19:
i currently only have a esp32 but ill probably have to buy the cheapest thing that will do the job

LeeNX, 2024-03-19:
Wired as in USB? Not all ESP32 device can do USB HID. Same for Arduino devices. There are some hacks, but they don't always work. For ease of use for a beginner, I would start with the GP2040 project. Lots of information and helpful community.  
  
We can close this issue?

Grandmaquil, 2024-03-19:
alr thanks we can close it

closed 2024-03-19 by Grandmaquil