lemmingDev/ESP32-BLE-Gamepad #215: GamePad with 2 battery level

Gabry11, 2024-03-27:
hi, it's possible to view 2 battery level info like a couple of ear buds?

lemmingDev, 2024-03-28:
Don't really see a use case
What would you use it for?

Gabry11, 2024-03-28:
I'm creating two twin gamepads, inspired by the old Nintendo Wii controllers. Each gamepad, right and left, has its own dedicated battery and processor. Therefore, the controllers communicate with each other, but only one connects to the PC via Bluetooth Low Energy (BLE), simulating the inputs of both controllers. I would like to be able to monitor the battery level of both gamepads, similar to how it's done with Bluetooth in-ear headphones.

LeeNX, 2024-09-14:
@Gabry11 can you see both earbuds in something like the Bluetooth controller and each battery level independent of each other? Did you need to install software to get this info? I don't think I have seen any HID descriptors for additional batteries.  
  
Maybe if you got the ```HID report descriptors```, we could see how it was done.

LeeNX, 2024-09-15:
Was thinking about how would you expect this info to be present by the host OS? We can add almost any services and/or characteristic, but most host OSs might not present this info to you, then reason I ask if you had seen this with an App or if your host OS did display this info and what it would look like. I could see how this could be present on Android with something like the  ```Bat ON``` App, which now looks like it has been abandoned, bugger. Here is a video - [BatON for Android](https://youtu.be/8YjL-Y_eZp8?si=8f-QR6OuYf_ioUt1) 
  
I could see how that might be useful on something like RetroPi or something, but now I am getting a little carried away.

lemmingDev, 2025-01-29:
https://h2zero.github.io/NimBLE-Arduino/class_nim_b_l_e_h_i_d_device.html#ab24968002ba94c2029e9d55364e1ec2b

NimBLE is set to have only one battery by the looksof it, and that's the method we currently use to set the level. Maybe you could ask about it over there

Maybe another one can be added our end, but i don't have much interest

FWIW - It seems a HID Descriptor can be set to have left and right levels with:

```
    // Battery level for left side
    0x05, 0x0F,        // USAGE_PAGE (Battery System)
    0x09, 0x01,        // USAGE (Battery Strength)
    0x85, 0x02,        // REPORT_ID (2) for Left Battery
    0x75, 0x08,        // REPORT_SIZE (8)
    0x95, 0x01,        // REPORT_COUNT (1)
    0x81, 0x02,        // INPUT (Data, Var, Abs)

    // Battery level for right side
    0x05, 0x0F,        // USAGE_PAGE (Battery System)
    0x09, 0x02,        // USAGE (Battery Strength)
    0x85, 0x03,        // REPORT_ID (3) for Right Battery
    0x75, 0x08,        // REPORT_SIZE (8)
    0x95, 0x01,        // REPORT_COUNT (1)
    0x81, 0x02,        // INPUT (Data, Var, Abs)
```

lemmingDev, 2025-01-30:
@afpineda Thanks for looking into that

Perhaps the easiest method @Gabry11 is to just send the left battery level every 30 seconds and the right battery level every 30 seconds 

No adjustments needed to the library and will work on all OS

lemmingDev, 2025-02-07:
Might close this...

closed 2025-02-07 by lemmingDev

LeeNX, 2025-11-17:
Ran across this article today - [Linux Proposal Aims To Overcome Kernel Limitation Affecting Various Gaming Peripherals](https://www.phoronix.com/news/Linux-Multi-Battery-HID-Devices)

And had a crazy idea of a gamepad with hot-swappable batteries, maybe two batteries in a device, where you might have a marathon gaming session and need to swap a flat battery for a fresh battery without turning off your gamepad.

Anyways, Linux might support multiple batteries soon, don't know about other OSs.