lemmingDev/ESP32-BLE-Gamepad #217: issues with the name reporting

benbeezy, 2024-04-09:
when doing something like `BleGamepad bleGamepad("aaa", "aaa", 100); ` or even trying to use the default I see Unknown gamepad on this hardware test site https://hardwaretester.com/gamepad and in unity I get the name showing up as "'ࠁ ܀ש̔ (HID)" but every time I connect to the device it shows up as something completely different but always some random junk data. I am using a doit esp32 devkit 1 board. Everything is working fine in windows and showing the right name, but in other applications it shows the random name.

Here is a screen shot in unity of what the device shows up as.
[image]

benbeezy, 2024-04-09:
I downgraded to 0.4.4 and the name reports much more normal. It seems to be a string that looks like D820B5A6 but at least now it is more consistent but I can't seem to change that name and no longer have access to things like setFirmwareRevision or setSerialNumber

benbeezy, 2024-04-11:
I have deleted and re-added the device each time. looks like I was wrong about 0.4.4 working as intended, I had just also rebooted unity at the same time and after a unity reboot it seems okay until I re-pair while unity is still up. All other devices seem fine with a mid run connection but for some reason not my esp32

Looks like if I reboot unity after the pairing then it shows up more normal but for whatever reason it's coming up as an ASRock A02019100900 and on another machine it's coming up as a Texas instruments device of some kind. I think this might be a larger issue on the unity side of things and less of an issue with this library.

lemmingDev, 2024-04-16:
Perhaps if you customise the VID and PID, it will realise its a different board

See the CharacteristicsConfiguration example on how to do this

LeeNX, 2024-07-10:
I would use a tools like [nRF Toolbox for Bluetooth LE](https://play.google.com/store/apps/details?id=no.nordicsemi.android.nrftoolbox)
  
Which would let you interrogate BLE characteristic & attributes, with very little OS interface. Hope that helps.

lemmingDev, 2025-01-19:
Did you still need help with this?

Not updated in 6 months...

Chris-656, 2025-01-23:
I tested with the new version with the example BLE characteristic & attributes. 
I tried to show peoperties with joy.cpl under windows 11. The device name is not shown correctly

[image]

Also tested the device manger view on the device. The name here is correct, but i cannot read or see the properties given in the example.

```
bleGamepadConfig.setModelNumber("1.0");
bleGamepadConfig.setSoftwareRevision("Software Rev 1");
bleGamepadConfig.setSerialNumber("9876543210");
bleGamepadConfig.setFirmwareRevision("2.0");
bleGamepadConfig.setHardwareRevision("1.7");
```
best regards Chris

Chris-656, 2025-01-25:
Thx for fast answer.
I tested now a lot on that issue.

First of all I used VId and PID definition to distinguish my gamepad.
```
bleGamepadConfig.setVid(0xe502);
bleGamepadConfig.setPid(0xeb37);
```

I used the link to test the gamepad in MS Edge (https://hardwaretester.com/gamepad) 

[image]

I see the gamepad but no name. In the registry the key only appears or is created when I start joy.cpl on the command line or use the joystick class to read values in a C# programm.

[image]

So changing the name you propose is working then it appears in the Gaming App

[image]

[image]
So maybe the reg file can be hard coded to rename it when the vid and pid is not changing,

regards Chris

lemmingDev, 2025-01-26:
Glad it's working for you

We might need to add the instructions I gave to a FAQ

lemmingDev, 2025-01-26:
You can to to get an official vid and pid for free at https://pid.codes/

I would assume it would then work without registry editing etc

lemmingDev, 2025-01-29:
Will close as it seems to be resolved. Feel free to comment if you need more help

closed 2025-01-29 by lemmingDev