lemmingDev/ESP32-BLE-Gamepad #218: Analog triggers on Android

WIrral, 2024-04-23:
Hi

I've been trying to send a pot value to android as the right trigger value using the example for the pot axis.

The provided example works perfectly as is, but if i change the X axis to any of the trigger axes, eg. bleGamepad.setRY the device doesn't recognize any input.

On windows this works just fine, but i can't for the life of me figure out what axis the android device expects for the right trigger..

Thanks

Yangzepeng5382, 2024-04-25:
Maybe you changed the key to a pin that does not have ADC functionality

WIrral, 2024-04-25:
> Maybe you changed the key to a pin that does not have ADC functionality

Unfortunately not.
All other axes can be used, it's just that Android doesn't recognise the RY input as a trigger input, and I don't know what it would recognise as a trigger input.

WIrral, 2024-05-11:
For anyone who might have the same issue, I figured it out.

I ended up checking using a real bt controller and a gamepad tester app and noticed that android calls the right trigger "GAS".

So, to set an analog value on a trigger for android, you can enable the simulation controls and use bleGamepad.setAccelerator() and andoid will read this as an analog right trigger input. I haven't tested, but i assume bleGamepad.setBrake() would do the same for the left trigger.

closed 2024-05-13 by WIrral

WIrral, 2024-05-13:
> @WIrral why not make a PR with your findings as helpful direction as documentation, than having to search the issue for extra info?

I'd love to do that, but i'm entirely new to Github and have no idea how any of this works :) 
I'm also not certain if this is the intended solution or a workaround. My actual knowledge of coding is extremely limited unfortunately

LeeNX, 2024-05-13:
@WIrral you already have done quite a bit, don't think any less of your contribution, but if you up to the task, then just one more step is even more progress.  
  
Fork the repo, make some changes to the ```README.md``` in your repo, that outlines your findings. HID usage does not have a fixed standard. It's open to the host OS and how it's used. This is why Android usage is a little different.  
  
If you make a note about how the Android implementation is different, would help the next user on Android or the next user that finds a different implementation for a different host OS.  
  
Once you happy with the update, submit a Pull Request (PR) to the main repo for review and change suggestions.  
  
Rinse and repeat and now you have a new achievement unlocked, this projects has a little more easier to access info and hopeful get the ball rolling faster for others. I hope this helps and also nudges others to add their little bit of help.

WIrral, 2024-05-13:
@LeeNX Thanks for the info, I've gone ahead and done that!

LeeNX, 2024-05-14:
@WIrral that looks perfect, but let's see what @lemmingDev says.  
  
Maybe for completeness, what Android version was this tested on, maybe what device, just in case it has some sort of custom firmware and what tool on the Android device did you make the test with?  
  
I have personally used ```Game Controller Tester - v1.45```, but I have not tested the triggers.

WIrral, 2024-05-14:
@LeeNX The phone I ultimately used it on (a samsung galaxy a40) is currently in use in an "art installation" that the BT control was used for, will update on the android version when I get it back.

But it also worked fine on my OnePlus running OxygenOS 11.0.16 (android 11).

I used `Gamepad tester` published on google play store by `elron`. It is version `23.08.18`.