lemmingDev/ESP32-BLE-Gamepad #219: Rocker zero float

Yangzepeng5382, 2024-04-25:
When I use the joystick sometimes have some float, you can add some filtering algorithms, before I did a low-pass filter, but did not have a good effect, it is best to add Kalman filter this effect is still quite considerable.It would be nice to add a gyroscope function.

LeeNX, 2024-07-10:
Oversampling, filtering and/or averaging generally would add some sort of lag and resources usage. This might be needed, but I am not sure this is suited for the library, but maybe a good working example would be nice. Something like the debounce example - https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/MultipleButtonsDebounce/MultipleButtonsDebounce.ino 
  
If you come up with some working, let me know.

lemmingDev, 2025-01-12:
I think LeeNX is right

Either use a premade solution such as Bounce2 used in the given example, or your own implementation as needed.

Shouldn't really include it in this library

As for gyroscope, there is an existing issue here https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/141

Will close

closed 2025-01-12 by lemmingDev