lemmingDev/ESP32-BLE-Gamepad #220: 'NimBLEHIDDevice' does not name a type; did you mean 'NimBLEDevice'?

IanClayman, 2024-04-28:
Getting this message when trying to verify a very basic sketch that just initialized a BleGamepad:

`In file included from C:\Users\---\AppData\Local\Temp\.arduinoIDE-unsaved2024327-23876-1d2cf2o.t5zo\sketch_apr27a\sketch_apr27a.ino:2:
c:\Users\---\Documents\Arduino\libraries\ESP32-BLE-Gamepad/BleGamepad.h:41:5: error: 'NimBLEHIDDevice' does not name a type; did you mean 'NimBLEDevice'?
     NimBLEHIDDevice *hid;
     ^~~~~~~~~~~~~~~
     NimBLEDevice

exit status 1`

Board: Xiao ESP32C3
Please let me know if I need to provide any other information

LeeNX, 2025-01-04:
@IanClayman up for testing new release and letting us if all is good and if so closing this issue? Thanks.

lemmingDev, 2025-01-12:
This is fixed when using latest version of this library and lastest NimBLE

closed 2025-01-12 by lemmingDev