lemmingDev/ESP32-BLE-Gamepad #222: Naming the device

ernest-cpu, 2024-06-10:
Is there a way to name the device in game controllers menu? At the moment it's called based on how many inputs you have. I would like to change it. I was able to change the way it called as a bluetooth device in device manager without any issue.
[image]

LeeNX, 2024-06-10:
Have you tried - https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/11ee6e927df7c8bef37b2cdc36eb798e5173c15f/examples/CharacteristicsConfiguration/CharacteristicsConfiguration.ino#L10

Remember to delete/forget your BLE device before re-pairing, else the cached characteristics can causes problems.

ernest-cpu, 2024-06-10:
That is not my issue. I can change the name it shows up in windows with device manager and when pairing without any issue. My issue that the name it shows up with in the game controllers window is not the same as everywhere else, it calls itself the number of buttons and axis I'm using automatically so if I where to start using 2 axis then it would call itself a 2 axis 28 button device. I repair it every time and even go as far as to reset the vids and pids so I can see the new changes so I know that's not the issue.

LeeNX, 2024-06-10:
Do other gamepad/controllers work as expected or is this an ```undocumented``` feature of the  ```Game Controllers``` applet?  
  
BLE implementation and interface is very much up to the Host OS. I don't believe that there is anything that could be done in this library to change that, but I have not used Windows much in this regard, maybe somebody else has a different feedback.

lemmingDev, 2024-06-10:
Changing it can be done:

1.  Open regedit and navigate to `Computer\HKEY_CURRENT_USER\System\CurrentControlSet\Control\MediaProperties\PrivateProperties\Joystick\OEM\`

2. Click on the VID/PID folder you'd like to change in the left pane

3. Double click on the OEMName icon in the right pane and set the name you like

4. Marvel at your new gamepad name

[image]

LeeNX, 2024-06-11:
Possible something that could be put into ```SomeJoy.inf``` file or something?

ernest-cpu, 2024-06-11:
when I use a usb gamepad the name it shows up with in windows matches up to the name of the gamepad device. not sure why this is different with bluetooth. or how do other bluetooth controllers have the correct name show up in game controllers menu. maybe the esp ble gamepad is not reporting a name for that and windows is guessing?

LeeNX, 2024-06-12:
Disclaimer, it's been a long time since I have used Windows and things could have changed alot, take my feedback with a grain of salt.  
  
The USB based gamepad you used, were they big brands or did you have to install drivers/inf files?

Windows (and other OS) has pre existing drivers and information setup for bigger or common products. These are normally tied to the Vendor ID / Product ID exposed by PCI/USB/BLE interface, which needs to be registered. Or can be updated with oem.inf files, which can provide more info and driver tooling. I believe that Windows is using a generic naming scheme based on what is exposed in the gamepad characteristics.  
  
It is possible that ```ESP32-BLE-Gamepad``` library is not exposing something that Windows would rather use as gamepad name, but unless somebody knows what this data or characteristics is, I don't see how we could improve this.  
  
You could fake your VID/PID to something that is known for a similar gamepad, but that could cause other problems and if you only worried about the display name, then ```oem.inf``` could be a work around.

ernest-cpu, 2024-06-13:
The usb gamepads used where of all different types, name brands and ones I made myself using https://github.com/MHeironimus/ArduinoJoystickLibrary for the pro micro, https://github.com/schnoog/Joystick_ESP32S2 or with teensy hid library on the teensy.   I never had to install any kind of drivers, they worked with the name I put in hid descriptors. I also tryed this with a Nintendo wiimote through bluetooth and the name that shows up is correct even though I never installed any drivers for it.

ernest-cpu, 2024-07-23:
https://www.bluetooth.com/specifications/specs/human-interface-device-profile-1-1-1/  
5.3.3 SDP Attribute Summary: Service Name
This attribute needs to be provided so that the hid device receives a name.

ernest-cpu, 2024-07-23:
https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/35

ernest-cpu, 2024-07-27:
I wasn't able to do it through bluetooth but by using the usb as hid also and using the same vid and pid as usb hid I was able to set it up so that if you plug the usb once into the computer it will remember the name of this device and once bluetooth connects it will appear with the same name. That way it should be pretty seamless.

closed 2024-07-27 by ernest-cpu