lemmingDev/ESP32-BLE-Gamepad #227: I cannot paired multiply devices on window 11

OatKungKTZ, 2024-08-28:
I can't connect to Bluetooth. This is my second wheel sim. I used the same code as my first wheel and just config VID/PID and serialnumber I try to forget the first wheel and this board can be paired but I have to unpair and pair the other wheel every time I want to switch my wheel sim. How can I fix it?
**Esp32 dev module**
[image]

LeeNX, 2024-09-14:
Similar to #225

lemmingDev, 2025-01-12:
Fixed in latest version https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.6.3

Tested with the exact same sketch, and I was able to independently verify in Game Controllers panel and a website tester
In both cases, input was working. Everything still worked after a reboot of both devices and Windows

Please confirm and close issue if all good

[image]

closed 2025-01-24 by lemmingDev