lemmingDev/ESP32-BLE-Gamepad #228: Multiple ESP32 to the same PC

Roigo21, 2024-09-14:
Hello, i have two ESP32's configured to act as gamepads, but i cant use both at the same time. On one, i only have VID and PID set, but on other i changed every parameter to try to communicate with both, and can only connect the other board if I unpair the first connected. 
Those are the parameters I set:
`bleGamepadConfig.setSerialNumber("9876543210"); 
    bleGamepadConfig.setModelNumber("123123");
    bleGamepadConfig.setHidReportId(12389332);
    bleGamepadConfig.setPid(0x503);
    bleGamepadConfig.setVid(0x503);
    bleGamepadConfig.setHardwareRevision("9876543210");
    bleGamepadConfig.setFirmwareRevision("9876543210");
    bleGamepadConfig.setSoftwareRevision("9876543210");`
This is the error I get (windows 11)
[image]

LeeNX, 2024-09-14:
Similar to #225

Roigo21, 2024-09-14:
> Similar to #225 

Yes i know, but the issue is still open. 
I found a workaround: using BLECompositeHID from @Mystfit

https://github.com/Mystfit/ESP32-BLE-CompositeHID
Right now i use BLECompositeHID in one board and BLEGamepad on other, and both connect.

LeeNX, 2024-09-14:
Opening another issue on the same topic splits the focus of other people keeping an eye on fixes or testing ideas.  I am not sure it's helpful.

Roigo21, 2024-09-14:
My bad, not used to issue creation on git

LeeNX, 2024-09-15:
No worries, this is how we learn. Happy you here and finding your way.  
  
Close this issue and add any info that might be useful on the first issue #225   
  
Hopefully we track down what is causing this issue and able to resolve the big problem.

microbmen, 2024-09-24:
Something changed on NimBLE-Arduino on version 1.4.2, 

  bleGamepadConfig.setVid(0xe502);
  bleGamepadConfig.setPid(0xabcd);

Doesn't work any longer.  No matter what its the same value 
   VID=0x02e5, PID=0xcdab
even if set to the above or anything else....

when i downgrade to NimBLE-Arduino 1.4.1 this works again.

OatKungKTZ, 2024-10-02:
You mean change VID/PID functions not work with NimBLE-Arduino version 1.4.2 but when you downgrade to 1.4.1 it's worked right?

microbmen, 2024-10-03:
> @microbmen 
> You mean change VID/PID functions not work with NimBLE-Arduino version 1.4.2 but when you downgrade to 1.4.1 it's worked right?

Correct

LeeNX, 2025-01-09:
Can you update and test with ESP32-BLE-Gamepad v0.6.1 and let us know if anything has improved?

LeeNX, 2025-01-09:
Thanks @Roigo21 , mind closing the issue, unless you wish to do more testing?

Roigo21, 2025-01-09:
> Thanks @Roigo21 , mind closing the issue, unless you wish to do more testing?

Yea that was what i was thinking, thanks for the help.

closed 2025-01-09 by Roigo21