lemmingDev/ESP32-BLE-Gamepad #229: setVid and setPid seems to be broken since NimBLE-Arduino 1.4.2

microbmen, 2024-09-24:
Something changed on NimBLE-Arduino on version 1.4.2, 

  bleGamepadConfig.setVid(0xe502);
  bleGamepadConfig.setPid(0xabcd);

Doesn't work any longer.  No matter what its the same value 
   VID=0x02e5, PID=0xcdab
even if set to the above or anything else....

when i downgrade to NimBLE-Arduino 1.4.1 this works again.

Mystfit, 2024-10-03:
Could this possibly be linked to this upstream [issue ](https://github.com/h2zero/NimBLE-Arduino/issues/492). The old behaviour was that there was some bit-shifting happening to the vid and pid values before they got stored in m_pnpCharacteristic. Looking BleGamepad::begin, I noticed that the high and low bytes for the vid and pid values are getting swapped before being passed to the pnp function call to set pnp values - maybe they values don't need to be swapped anymore?

Mystfit, 2024-10-03:
@microbmen Can confirm that getting rid of the high/low byte flipping that was occuring with the VID, PID and GUID values before they were sent to NimBLE fixed the problem for me. Seems that upstream they removed the need for those bytes to be flipped. Submitted a [PR](https://github.com/lemmingDev/ESP32-BLE-Gamepad/pull/230) to fix the issue.

closed 2024-12-17 by lemmingDev