lemmingDev/ESP32-BLE-Gamepad #231: alt, shift doesn't want to work

eluxking, 2024-10-07:
I try to use alt+tab to open multitasking and it doesn't work, and when I use tab+shift it only does tab, I want to know if anyone has been able to use it? What configuration should I add?

I have tried key_rightalt, key_leftalt, and it doesn't work.
#include <Arduino.h>
#include <KeyboardDevice.h>
#include <BleCompositeHID.h>

BleCompositeHID compositeHID("Titiritero", "Mystfit", 100);
KeyboardDevice* keyboard;

void setup()
{
    Serial.begin(115200);

    keyboard = new KeyboardDevice();
    compositeHID.addDevice(keyboard);
    compositeHID.begin();

    Serial.println("Esperando conexión...");
    delay(3000);
}
void loop{.....

....else if(command == "alt"){
                Serial.println("Presionando alt");
                keyboard->keyPress(KEY_RIGHTALT);
                delay(10);
            }else if(command == "alt-"){
                Serial.println("Soltando alt");
                keyboard->keyRelease(KEY_RIGHTALT);
                delay(10);
            }else ....
.....}

Mystfit, 2024-10-10:
I think this issue was meant for the ESP32-BLE-CompositeHID repo? Might want to close it here and re-open it there.

eluxking, 2024-10-16:
There won't be it there because for some reason the keyboard only works when my esp32 is configured as a gamebox.

I'm going to try the suggestion.

closed 2024-11-23 by lemmingDev