lemmingDev/ESP32-BLE-Gamepad #232: Can't custom configuration

fedekrum, 2024-10-18:
Hi

I am using an Intel Mac with Arduino IDE 2.3.3.
Libs are NimBLE-Arduino 1.4.1 by h2zero and ESP32-BLE-Gamepad 5.4
HW is a ESP-WROOM-32 Module
I have selected the "ESP32 Dev Module" ant the correct port.
I have no problem in compiling and uploading any sketch on it.

The problem comes when I try the example called ""CharacteristicsConfiguration.ino"
I have change the battery level to 50 %on the constructor.

It doesn't seems to change the name of the device

[image]

Any idea why is not changing?

Thanks in advance

lemmingDev, 2024-11-23:
You need to delete the controller from the operating system and re-add it if you make any changes

LeeNX, 2025-01-09:
@fedekrum did you ever get a chance to test further?

lemmingDev, 2025-01-22:
Setting a custom MAC address using the esp_base_mac_addr_set functionality doesn't seem to work anymore, so it will be removed from the example you tried

Issue will be closed, though please feel free to comment further or open another issue if needed