lemmingDev/ESP32-BLE-Gamepad #235: Games on Mac and Windows cannot recognize ble controller inputs

JasonARong, 2024-11-23:
I tried to create an esp controller that functions similar to an xbox controller. It is able to connect to Mac, PC and Android, and on Android, it works fine when I tested it with GamePad Tester. However, on pc & mac even though the inputs are able to be picked up by websites like [Hardware Tester](https://hardwaretester.com/gamepad) and [Controller Tester](https://www.onlinemictest.com/controller-tester/), in some games the inputs are registered as generic controller inputs like(btn0, btn1, axis0, etc) so I have to use the in-game controller remapper to set up the control scheme, and in some other games the esp controller will be completely ignored. 

One thing I noticed is that for official controllers it will say standard mapping on [Hardware Tester](https://hardwaretester.com/gamepad) and show an illustration of the controller, so is it possible to somehow trick the system to think my esp controller is a standard controller?

I have been struggling with this for a while, please help me 😭

LeeNX, 2024-11-23:
If you looking to mimic an XBox controller, then it might be better to look at Mystfit's library https://github.com/Mystfit/ESP32-BLE-CompositeHID
  
The ESP32-BLE-Gamepad library is just a BLE gamepad, using the BlueTooth standards interface and up to the Host OS to interrupt.  
  
Many of the Testing App's that show a controller outline or picture are matching on the VID:PID of the reported device. You could fake the VID:PID of the device to look the same as the device you want to report, but that might cause other funnies and generally not a perfect solution.  
  
I hope you find what you looking for.

lemmingDev, 2024-11-23:
> If you looking to mimic an XBox controller, then it might be better to look at Mystfit's library https://github.com/Mystfit/ESP32-BLE-CompositeHID
>
> The ESP32-BLE-Gamepad library is just a BLE [...]

Ditto

A lot of games require xinput, and the above mentioned library will help you with that

closed 2024-11-23 by JasonARong