lemmingDev/ESP32-BLE-Gamepad #237: bleGamepad.isConnected() aways return false

emanuel-xavier, 2024-11-30:
Hi,

I'm trying to make setup some buttons, with ESP32-BLE-Gamepad but it's not working.

My code looks like this:
`c++

 bleGamepad.begin();
  delay(2500);
  Serial.println("Setup done");

if (bleGamepad.isConnected()) {
    .
    .
    .
} else {
    Serial.println("BLE Gamepad is not connected");
  }
`

My computer is connected to the device, as you can see in the image below:
[image]

It appears as a game controller:
[image]

But it keeps printing "BLE Gamepad is not connected" on the serial monitor. bleGamepad.isConnected() is always returning false.

I'm using an ESP32-WROOM-32, and the ESP-BLE-Gamepad version is 0.5.4.

LeeNX, 2024-12-01:
@emanuel-xavier your code snippet does not give very much info to go on, so here are some suggestions that are pretty battle proven.  
  
Have you tried any of the examples unaltered. I like to start with [SingleButton](https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/SingleButton/SingleButton.inourl) sketch, thou I am not a fan of ```BUTTON_1``` and it also does not run without an actually button pressing thing. Plus on my mobile device, it presses things while I am testing that can be a pain.
  
My next goto is [SetBatteryLevel](https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/SetBatteryLevel/SetBatteryLevel.ino) for debugging.  
  
Let us know how your testing goes. Best of luck.

emanuel-xavier, 2024-12-01:
Hi guys, thanks for the support!

[Here](https://github.com/emanuel-xavier/ESP32-steering-wheel-button-box/blob/main/ec2-button-box.ino) is the code I'm trying to use.

I also tried the single button example before and encountered the same problem.

I'll try using SetBatteryLevel and check what happens.

Sab1e-GitHub, 2024-12-02:
Hi @emanuel-xavier ,After my testing, I found that your code itself has no issues, but it seems that the problem lies in the update of NimBLE library to version 2.0.0, where many methods were overridden, causing bleGamepad.isConnected() to always return false. I'm currently working on fixing these issues. If you're using the latest version (2.0.0) of NimBLE library, and need to use Esp32-BLE-Gamepad library urgently, you can try downgrading NimBLE to version 1.4.3 for now, which should resolve your issue!  

Thank you so much for raising this Issue!

emanuel-xavier, 2024-12-03:
@Sab1e-GitHub, thank you so much! I downgraded ESP32-BLE-Gamepad from 0.6.0 to 0.5.4 and NimBLE-Arduino to 1.4.1, and now it's working.

closed 2024-12-03 by emanuel-xavier