lemmingDev/ESP32-BLE-Gamepad #239: bluetooth does not appear on the pc only on the cell phone

Mrbraule, 2024-12-04:
After several attempts to make the connection in the matrix where only 4 buttons worked, I tried to do the programming again, after which I was unable to communicate with the computer.

Mrbraule, 2024-12-06:
> @Mrbrauleisso está no Windows por acaso? Você pode confirmar o SO e que sua interface BlueTooth no seu SO host é compatível com BLE?
> 
> Se você tiver um ESP32S (clássico), pode testar a conexão com seu sistema operacional host para confirmar se o bluetooth está funcionando como uma primeira etapa de depuração.

Sure it is compatible, yes, as it worked previously, but only 4 buttons worked, I was following a tutorial and in the tutorial it showed 32 buttons and on mine only 16, after I tried to do new programming it started to look like this, I'm using a BT adapter or Baseus

LeeNX, 2024-12-06:
Not used mail reply, please excuse any crazy.

Have you removed /deleted /unpaired the old device on the host OS and try
re-pairing.

Host OS are known to cache the ble setup and changes you make while
testing, like new buttons do not get picked up and even break device usage.

Best of luck

Mrbraule, 2024-12-06:
> Não usei resposta de e-mail, desculpe qualquer loucura. Você removeu/excluiu/desemparelhado o dispositivo antigo no sistema operacional host e tentou emparelhar novamente. Os sistemas operacionais host são conhecidos por armazenar em cache a configuração ble e as alterações que você faz durante o teste, como novos botões não são selecionados e até interrompem o uso do dispositivo. Boa sorte
> […](#)

I can't fix the old pairing error, today I'm going to test the adapter with a PS4 controller and if it works I'll see if I can remove the old pairing from the records

LeeNX, 2024-12-07:
Good to see that you making progress.  
  
You can take a look at https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/MultipleButtons/MultipleButtons.ino, which has 10 buttons. Extend and test as you add more buttons/pins. Documented max limit for buttons is 128, thou I am guessing that is not directly using GPIOs.  
 
Remember to unpair your old device on the host OS between changes, else you will run into the same breaking experience.   
  
Going to 32 buttons might need something like a i2c multiplexer as I don't know of an ESP32 device that has that many free GPIO pins, as an example - https://randomnerdtutorials.com/esp32-pinout-reference-gpios/ with some info of which pins might be usable.  
  
I have run into the problem where using a pin as an LED (the wrong PIN - GPIO8), halted the start of my ESP32 test project, thou flashing an already known working project/firmware worked. My problem was because I was using two difference ESP32 dev boards with different PIN assignments and usage. Check your ESP32 board documentation.  
  
Let us know how your project goes. Best of luck.

Mrbraule, 2024-12-07:
> É bom ver que você está progredindo.
> 
> Você pode dar uma olhada em https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/MultipleButtons/MultipleButtons.ino , que tem 10 botões. Estenda e teste conforme você adiciona mais botões/pinos. O limite máximo documentado para botões é 128, embora eu esteja supondo que não esteja usando GPIOs diretamente.
> 
> Lembre-se de desemparelhar seu dispositivo antigo no sistema operacional host entre as alterações, caso contrário, você terá a mesma experiência de quebra.
> 
> Ir para 32 botões pode precisar de algo como um multiplexador i2c, pois não conheço nenhum dispositivo ESP32 que tenha tantos pinos GPIO livres, por exemplo - https://randomnerdtutorials.com/esp32-pinout-reference-gpios/ com algumas informações sobre quais pinos podem ser utilizáveis.
> 
> Eu me deparei com o problema em que usar um pino como um LED (o PIN errado - GPIO8), interrompeu o início do meu projeto de teste ESP32, embora piscar um projeto/firmware já conhecido funcionasse. Meu problema era porque eu estava usando duas placas de desenvolvimento ESP32 diferentes com atribuições de PIN e uso diferentes. Verifique a documentação da sua placa ESP32.
> 
> Conte-nos como foi seu projeto. Boa sorte.

I just tried programming a tutorial that I follow and it didn't work https://www.youtube.com/watch?v=sDdLMyt--Sw&lc=Ugx-gMjAXfYNO0l5-HR4AaABAg.ABRCBlTuY35ABUj_woPu9G

I don't do any programming, I just wanted to put buttons on my homemade steering wheel,
This programming that you mentioned with 10 buttons is just creating a code and placing it??
Do you have Discord?

LeeNX, 2024-12-07:
Sorry, I have not seen the tutorial and the library version is pretty old v0.1.9, if I read the YT description correctly.  
  
The sketch suggested is one of the examples bundled with the ESP32-BLE-Gaempad library in Arduino IDE.  
  
As for should it run without any changes, it should within limitation of what GPIO pins are used by your ESP32 board.  
  
The buttons are mapped to GPIOs around
https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/4ffb1d429bf9c006dc94837ee2c36fa3f52a38f5/examples/MultipleButtons/MultipleButtons.ino#L19

Again, remember to unpair old device between changes when testing BLE updates and changes.  
  
Sorry, no Discord that I know of.

LeeNX, 2025-01-09:
@Mrbraule did you make progress?

lemmingDev, 2025-01-22:
I will close as there has been no response in 2 weeks

closed 2025-01-22 by lemmingDev