lemmingDev/ESP32-BLE-Gamepad #240: Another 5.x release with new features?

LeeNX, 2024-12-06:
Please can we get a 5.x release still using [NimBLE-Arduino](https://github.com/h2zero/NimBLE-Arduino) 1.4.3?  
  
I really would like to test the OutPut feature, but currently can't unless I cut my own release in my fork. Looking to test with PlatformIO. I think a really cool test would be sending text to a screen on the gamepad.

Sab1e-GitHub, 2024-12-06:
I think if you don't upgrade NimBLE-Arduino to version 2.0.0, the 5.4 release should still work fine with all features.

LeeNX, 2024-12-06:
I will test later, but I think I did and my compile broke, but I don't have that info handy right now. I will report back over the weekend.

LeeNX, 2024-12-07:
Thanks @Sab1e-GitHub ... But I think there is a bunch of confusion over NimBLE-Arduino v2.0.0. The link for  NimBLE-Arduino v2.0.0 in the ESP-32-BLE-Gamepad README.md is 404 and I came across post about how h2zero might have mixed up some release parts. I am not sure what branch or tag might actually be NimBLE-Arduino v2.0.0 to target to download or test with.  
  
I was just hoping we could possible have a release between 6.0 and 5.4 which would include your updates with Output reports.  
  
I did test a basic sketch with ESP32-BLE-Gamepad v5.4 and NimBLE-Arduino v1.4.3 and it seems the error I was running into was the missing comma, which I fixed locally.  
  
Really enjoying PlatformIO with VS Code as a working environment, which means I can switch Arduino libraries and releases per project and even make fixes in the local library like I did above.  
  
Still going to ask, no, beg for a v5.5 release until NimBLE-Arduino v2.0.0 is public released and possible supported by PlatformIO ... because I am lazy ... Maybe I should just make the v5.5 and send a PR.

Sab1e-GitHub, 2024-12-07:
You're welcome @LeeNX 

First, regarding the issue with NimBLE-Arduino v2.0.0, it seems that h2zero hasn’t released a 2.0.0 tag on GitHub. However, he has updated the 2.0.0 version in the Arduino Library Manager, which is indeed confusing. If you prefer not to use the Arduino Library Manager to get the latest library, you can directly clone his [repository](https://github.com/h2zero/NimBLE-Arduino.git).

Additionally, I apologize for misunderstanding your request. I forgot that we haven’t released a 5.x version with the Output Report functionality. If you specifically need the 5.4 version with Output functionality, you can use the following command to roll back to a previous commit cdf05b3fccdee7fb5546fb3b0262f6c91e1b0efa:

`git reset --hard cdf05b3fccdee7fb5546fb3b0262f6c91e1b0efa`

This is the simplest method. If you indeed want to create and submit a Pull Request, you can roll back to this commit and proceed from there.

LeeNX, 2024-12-07:
Haha, thanks @Sab1e-GitHub you really enabling my lazy practices.  
  
Also, not need for apologises, we all working together to make things better.  
  
Took me a few different approaches until I found a way to work with this. Using PlatformIO, I was able to pull this commit hash and also make a long fix - https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/cdf05b3fccdee7fb5546fb3b0262f6c91e1b0efa/BleGamepadConfiguration.cpp#L22. 
Needing a comma at the end. More for anybody else that lands here.  
  
My platformio.ini
```
env:lolin_c3_mini]
platform = espressif32
board = lolin_c3_mini
; change microcontroller
board_build.mcu = esp32c3
; change MCU frequency
#board_build.f_cpu = 160000000L
framework = arduino
build_flags = 
  -D ARDUINO_USB_MODE=1
  -D ARDUINO_USB_CDC_ON_BOOT=1
  -D CORE_DEBUG_LEVEL=5
monitor_speed = 115200
lib_deps = 
	#lemmingdev/ESP32-BLE-Gamepad@^0.5.4
        https://github.com/lemmingDev/ESP32-BLE-Gamepad.git#cdf05b3fccdee7fb5546fb3b0262f6c91e1b0efa
	h2zero/NimBLE-Arduino@^1.4.3
```

My needs are currently covered and going to play and test further, but for ease of use and possibly other users of the library, might be worth an intermediate release that is working with little effort as possible.

Sab1e-GitHub, 2024-12-07:
@LeeNX I wasn't sure how to create a Pull Request for an older version,😂 so I published a v5.6 Release in my fork. You can find it [here](https://github.com/Sab1e-GitHub/ESP32-BLE-Gamepad/releases/tag/v0.5.6).

LeeNX, 2024-12-07:
Now you just showing off ... hahah ... ;-)

Thanks @Sab1e-GitHub ,  ... I have not even got to work through my current tests.   
  
Mmm, that is an interesting idea, never have I thought about a PR for a downgrade/earlier release. I am guessing make a branch and hopefully @lemmingDev could sync that branch into the main repo and then make a release from there, without making the latest or something ... but now this thread is getting way off track.  
  
I and possible other people now have multiple ways to poke at this. Let's see where it

LeeNX, 2024-12-07:
Just for completeness, I was able to compile the test sketch using @Sab1e-GitHub release for v5.6, thanks @Sab1e-GitHub .

project platformio.ini
```
[env:lolin_c3_mini]
platform = espressif32
board = lolin_c3_mini
; change microcontroller
board_build.mcu = esp32c3
; change MCU frequency
#board_build.f_cpu = 160000000L
framework = arduino
build_flags = 
  -D ARDUINO_USB_MODE=1
  -D ARDUINO_USB_CDC_ON_BOOT=1
  -D CORE_DEBUG_LEVEL=5
monitor_speed = 115200
lib_deps = 
	# lemmingdev/ESP32-BLE-Gamepad@^0.5.4
        # https://github.com/lemmingDev/ESP32-BLE-Gamepad.git#cdf05b3fccdee7fb5546fb3b0262f6c91e1b0efa
        https://github.com/Sab1e-GitHub/ESP32-BLE-Gamepad.git#v0.5.6
	h2zero/NimBLE-Arduino@^1.4.3
```
Now to test more than just compile, need to flash this and do real world testing.

Sab1e-GitHub, 2024-12-07:
This is really amazing, @LeeNX ! 
If you have any other questions, feel free to ask me, and I will do my best to help!

Sab1e-GitHub, 2024-12-07:
Hi @LeeNX 
Due to someone accidentally deleting my code, the Output Report functionality was not working properly. I have submitted a [pull request](https://github.com/lemmingDev/ESP32-BLE-Gamepad/pull/241) to fix it, and I have also updated the old version you needed, which you can check [here](https://github.com/Sab1e-GitHub/ESP32-BLE-Gamepad/releases/tag/v0.5.6.1).  
I have tested the functionality, and everything is working fine. I have also updated the testing tool, which you can find [here](https://github.com/Sab1e-GitHub/HIDOutputDemo_CSharp).

LeeNX, 2024-12-07:
@Sab1e-GitHub  I hate it when elves eat my homework ;-)  

Thanks for the heads up, I will update my project and test further. I had called it a day. Will pick up my testing tomorrow.  
  
I am hoping I can come up with a python script to show off this feature, but it might take a few days.

lemmingDev, 2024-12-09:
Hi All 

Let me know what changes I should make to best sort this out

The NimBLE v2.0.0 release without actual link on GitHub really threw a spanner in the works

Sab1e-GitHub, 2024-12-10:
Hi @lemmingDev ,
I don't think we have a perfect solution to this issue. The best approach would be for h2zero to release version 2.0.0. You can create an issue on his repository to remind him.

If you want this repository to support both NimBLE Arduino 1.4.3 and 2.0.0, you could try adding precompiled conditional statements to check the version and adapt the code accordingly.

The code might look like this:

```
#ifdef LIB_VERSION_1_4_3
    void A(int x) ;
#elif defined(LIB_VERSION_2_0_0)
    void A(double x) ;
#else
    #error "Unsupported library version"
#endif
```

However, I believe the best solution is still to have the user update NimBLE to version 2.0.0.

LeeNX, 2025-01-04:
I think this is resolved in general, so we can close. Thanks everybody.

closed 2025-01-04 by LeeNX