lemmingDev/ESP32-BLE-Gamepad #242: Setting axis only works for X, Y, RX, RY (Axis 0, 1, 3, 4)

JasonARong, 2024-12-07:
```
    int xValue = map(analogRead(joystickXPin), 0, JOYSTICK_X_MAX, 32767, 0);
    int yValue = map(analogRead(joystickYPin), 0, JOYSTICK_Y_MAX, 32767, 0);
    bleGamepad.setLeftThumb(xValue, yValue); // Works
    bleGamepad.setX(xValue); // Works
    bleGamepad.setY(yValue); // Works
    bleGamepad.setZ(xValue); // Does not Work
    bleGamepad.setRightThumb(xValue, yValue); // Works
    bleGamepad.setRX(xValue); // Works
    bleGamepad.setRY(yValue); // Works
    bleGamepad.setRZ(xValue); // Does not Work
    bleGamepad.setSlider(xValue); // This and all other sliders do not Work
```

LeeNX, 2024-12-07:
Have you tested the include example IndividualAxes.ino or even TestAll.ino? Remember to unpair / delete / remove bluetooth device from your host OS.

Sab1e-GitHub, 2024-12-07:
I used the following code for testing:
```
#include <Arduino.h>
#include <BleGamepad.h>

BleGamepad bleGamepad;

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  bleGamepad.begin();
}

void loop() {
  if (bleGamepad.isConnected()) {
    for (int i = 0; i < 32767; i += 10) {
      int xValue = i;
      int yValue = i;
      bleGamepad.setLeftThumb(xValue, yValue);   // Works
      bleGamepad.setX(xValue);                   // Works
      bleGamepad.setY(yValue);                   // Works
      bleGamepad.setZ(xValue);                   // Does not Work
      bleGamepad.setRightThumb(xValue, yValue);  // Works
      bleGamepad.setRX(xValue);                  // Works
      bleGamepad.setRY(yValue);                  // Works
      bleGamepad.setRZ(xValue);                  // Does not Work
      bleGamepad.setSlider(xValue);              // This and all other sliders do not Work
      delay(1);
    }
  }
}
```

I didn’t find any issues. Please take a look at the GIF below:
[image]
[image]

Could you provide more information?
(By the way, if you are using a custom configuration and have modified the number of buttons or axes, you must remove the device and re-pair it.)

JasonARong, 2024-12-07:
Thank you for your reply! Your testing code is also working for me. So it must have been me messing something up in my code.

Sab1e-GitHub, 2024-12-07:
You're welcome!  
Regarding custom configurations, you can find examples [here](https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/CharacteristicsConfiguration/CharacteristicsConfiguration.ino).

closed 2024-12-07 by JasonARong