lemmingDev/ESP32-BLE-Gamepad #243: Name of the controller in the game controller configuration window

JabberwockPL, 2024-12-07:
I have configured my device with the name and manufacturer and it is correctly displayed in the Windows Bluetooth interface; when I open the system configuration window for controllers, it does not have a specific name (Windows puts there a generic controller description). That is a bit of an issue, as some games identify the custom controller configuration by those names and if it is missing, the config is not saved. Am I doing something wrong or is the name not possible to configure? I do not even need my own name, it would be sufficient if it was hardcoded...

DrSimgear, 2024-12-09:
Check this, https://github.com/Mystfit/ESP32-BLE-CompositeHID/issues/10

JabberwockPL, 2024-12-09:
Indeed, this is the same issue, I have somehow missed it, sorry!

closed 2024-12-09 by JabberwockPL