lemmingDev/ESP32-BLE-Gamepad #244: No input from the device

Svetliin, 2024-12-08:
Hello,
I have an issue after uploading the example code. 
I am working on a 7 button Bluetooth gamepad and was trying the MultipleButtons.ino.

Code is uploaded successfully and the device is visible and can be connected on me PC (Windows OS).
Opening gamepad properties from Windows, I can see all buttons and axis, but when I press my button nothing is detected on the PC side.

My test button connects pin 35 and GND on the ESP32. Voltage drops from ~3.2 to 0 and from what I understood this should have triggered a button press on the PC.

Could you please share feedback, what could case this, as I tried 2 different ESP32 boards and the behavior is the same?

Note: I tried a different library (BLE-Keyboard) and that works perfectly, unfortunately for my DIY project, keyboard buttons are not suitable.

Sab1e-GitHub, 2024-12-08:
Hello @Svetliin 
You can first try removing your Bluetooth device and reconnecting it. If it still doesn't work properly, you can use the serial monitor to check whether your button presses are being detected.
If the issue persists, I hope you can provide more information. 
Good luck!

Svetliin, 2024-12-09:
Hello @Sab1e-GitHub,
Thanks for the response.

I remove the device from pairing each time I test a new code.
To test the serial monitor and exclude faulty hardware I used the below code with the coded button presses:

#include <Arduino.h>
#include <BleGamepad.h>

BleGamepad bleGamepad;

void setup()
{
    Serial.begin(115200);
    Serial.println("Starting BLE work!");
    bleGamepad.begin();
    // The default bleGamepad.begin() above enables 16 buttons, all axes, one hat, and no simulation controls or special buttons
    
}

void loop()
{
    if (bleGamepad.isConnected())
    {
        Serial.println("Press buttons 5, 16 and start. Move all enabled axes to max. Set DPAD (hat 1) to down right.");
        bleGamepad.press(BUTTON_5);
        bleGamepad.press(BUTTON_16);
        bleGamepad.pressStart();
        bleGamepad.setAxes(32767, 32767, 32767, 32767, 32767, 32767, 32767, 32767);
        bleGamepad.setHat1(HAT_DOWN_RIGHT);
        // All axes, sliders, hats etc can also be set independently. See the IndividualAxes.ino example
        delay(500);

        Serial.println("Release button 5 and start. Move all axes to min. Set DPAD (hat 1) to centred.");
        bleGamepad.release(BUTTON_5);
        bleGamepad.releaseStart();
        bleGamepad.setHat1(HAT_CENTERED);
        bleGamepad.setAxes(0, 0, 0, 0, 0, 0, 0, 0);
        delay(500);
           bleGamepad.sendReport();
    }
}

No inputs were detected in serial monitor and nothing was showing.

I will be happy to share all the details, but I am not sure what would be of help as this is the first time I am working with development boards.

Svetliin, 2024-12-10:
Hello,

I got it working.
It seems there was a software issue with Arduino IDE.
I re-installed it, installed the library and uploaded the same code and it worked.

Thank you for the help

closed 2024-12-10 by Svetliin