lemmingDev/ESP32-BLE-Gamepad #245: Cannot display the name of the game controller

sam-007-start, 2024-12-11:
I used an example of a gamepad inside and burned it onto an ESP32 S3 development board using Arduino. On the Win10 system, only one gamepad icon and input device were displayed, and the name of the gamepad was not displayed correctly. Moreover, the BLE debugger could not display the name, only N/A devices were displayed.

LeeNX, 2025-01-03:
Possible related to https://github.com/lemmingDev/ESP32-BLE-Gamepad/pull/250. 
  
Would be up for testing master branch which this was merged into and letting us know if that solved your problem?

LeeNX, 2025-01-06:
@sam-007-start new release out, mind testing and reporting back and possible closing the issue if all is good? Thanks.

lemmingDev, 2025-01-12:
@sam-007-start Please test with latest https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.6.3 and report back

lemmingDev, 2025-01-19:
First of all, use the latest version

Second, you need to change a registry entry if you want that name changed

**Custom display name by means of the Windows registry editor (Windows only)**

1. Open the "registry editor" (regedit.exe) with user privileges (not administrator).

2. Navigate to the following folder (registry key):
HKEY_CURRENT_USER\System\CurrentControlSet\Control\MediaProperties\PrivateProperties\Joystick\OEM

So, for the library default -> HKEY_CURRENT_USER\System\CurrentControlSet\Control\MediaProperties\PrivateProperties\Joystick\OEM\VID_E502&PID_BBAB

3. Inside that folder, locate the subfolder (registry key) for your device.

4. Those keys are named using this pattern: "VID -hex number- &PID -hex number- ", where "VID" means "vendor identifier" and "PID" means "product identifier". Together, they are a (plug-and-play) hardware identifier (ID). If your device uses BLE connectivity and the default hardware ID set in this project, that registry key should be "VID_1D50&PID_FFFF". 
 
5. Double-click (edit) the value "OEMName".

6. Set a custom display name and hit "enter".

lemmingDev, 2025-01-19:
Updated instruction above. Please test, report back, and close issue if resolved.

ivomarsan, 2025-01-23:
It works! But there isn't a way to auto configure that via code config?

lemmingDev, 2025-01-23:
Unfortunately not

closed 2025-01-23 by lemmingDev