lemmingDev/ESP32-BLE-Gamepad #246: PlatformIO error 'marked override'

CasperHxR47, 2024-12-12:
Hello, your library looks great! I'm getting the following error when using it for the first time today. 

Error:
BLEConnectionStatus.h
'void BleConnectionStatus::onConnect(NimBLEServer*, NimBLEConnInfo&)' marked 'override', but does not override [Ln 17, Col 10]
'void BleConnectionStatus::onDisconnect(NimBLEServer*, NimBLEConnInfo&)' marked 'override', but does not override [Ln 18, Col 10]

BLEOutputReceiver.h
'void BLEOutputReceiver::onWrite(NimBLECharacteristic*,NimBLEConnInfo&)' marked 'override', but does not override [Ln17, Col 10]

It seems that the Arduino IDE auto-includes a previous declaration if there isn't one. PlatformIO doesn't do that.

I removed the 'override' statement at the end of those 3 lines and recompiled without a problem.

Sab1e-GitHub, 2024-12-12:
Hello @CasperHxR47 ,  

Judging from the compilation errors, you need to manually install version 2.0.0 of NimBLE Arduino. Although NimBLE hasn't released version 2.0.0 on GitHub, you can directly clone their repository to get the latest version.  

For more information about this issue, you can check #240 .

LeeNX, 2024-12-12:
Could also clone the NimBLE Arduino repo, but that did not work for me.  
  
I updated my platformio.ini
```
lib_deps = 
  #lemmingdev/ESP32-BLE-Gamepad@^0.5.4
  # https://github.com/lemmingDev/ESP32-BLE-Gamepad.git#cdf05b3fccdee7fb5546fb3b0262f6c91e1b0efa
  https://github.com/Sab1e-GitHub/ESP32-BLE-Gamepad.git#v0.5.6
  h2zero/NimBLE-Arduino@^1.4.3
```

CasperHxR47, 2024-12-13:
h2zero/NimBLE-Arduino was just updated to 2.1.0! All is working now. Thank you all!

LeeNX, 2025-01-04:
If this is working, @CasperHxR47 please could you close the issue? Thanks.