lemmingDev/ESP32-BLE-Gamepad #248: Gamepad with IOS

Laurin08, 2024-12-22:
Is there still no workaround to use the gamepad with ios?
The BLE-Mouse library is supporting Ios so i think it should be possible somehow...

lemmingDev, 2024-12-23:
Try the xinput fork
https://github.com/Mystfit/ESP32-BLE-CompositeHID

and the xinput example
https://github.com/Mystfit/ESP32-BLE-CompositeHID/blob/master/examples/GamepadExamples/XboxXInputController/XboxXInputController.ino

Let us know how you go

Laurin08, 2024-12-23:
Hey 
Thank you for your answer!!
I would like to try it but I get several errors while compiling. I think the problem is the NimBLE library, with the older version 2.0.0 I got less errors but still something:

/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/GamepadDevice.cpp: In member function 'uint8_t GamepadDevice::specialButtonBitPosition(uint8_t)':
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/GamepadDevice.cpp:343:20: error: 'invalid_argument' is not a member of 'std'
  343 |         throw std::invalid_argument("Index out of range");
      |                    ^~~~~~~~~~~~~~~~
/Users/laurinmichel/Documents/Arduino/libraries/ESP32-BLE-CompositeHID/GamepadDevice.cpp:6:1: note: 'std::invalid_argument' is defined in header '<stdexcept>'; did you forget to '#include <stdexcept>'?
    5 | #include "esp32-hal-log.h"
  +++ |+#include <stdexcept>
    6 | #define LOG_TAG "GamepadDevice"

exit status 1

Compilation error: exit status 1

lemmingDev, 2024-12-23:
Might be best to ask on that specific page, though it might target NimBLE below 2.0.0

Laurin08, 2024-12-23:
Which nimble version do you use?

lemmingDev, 2024-12-24:
[NimBLE-Arduino-a79941cc9e77ed60136ee385b1c73d89c847bd94.zip](https://github.com/user-attachments/files/18242044/NimBLE-Arduino-a79941cc9e77ed60136ee385b1c73d89c847bd94.zip)

This is the version that the readme says to use

I just tried and it compiles fine with the xinput sketch, though I don't have an ESP32 handy to test with

Don't forget to also install the Callback library, as indicated in the readme
https://github.com/tomstewart89/Callback

lemmingDev, 2024-12-24:
Also, please report back whether or not it works with iOS

Laurin08, 2024-12-24:
Hi
Thanks again, I was able to clear the error with adding #include <stdexcept> in the GamepadDevice.cpp file in the Composite-HID library. now it works with the 2.0.0 version of nimble.
And it also works with IOS!

LeeNX, 2025-01-09:
@Laurin08 if all is good, do you mind closing the issue?

tangobravo, 2025-01-16:
Hi all,

I'm trying to figure out what iOS supports in terms of generic BLE gamepads and there is very little information out there. The iOS 16 release notes says:

tangobravo, 2025-01-16:
Hi all,

I'm trying to figure out what iOS supports in terms of generic BLE gamepads and there is very little information out there.

The [iOS 16 release notes](https://developer.apple.com/documentation/ios-ipados-release-notes/ios-16-release-notes#Game-Controller) say "Many additional Bluetooth and USB game controllers are supported by the Game Controller framework on macOS 13, iOS 16, and tvOS 16 and later. (82409809)". I'm hoping that means some sort of generic HID BLE gamepad support but it doesn't explicitly say that.

This issue, and your comment @Laurin08 "also it works on iOS", is the closest reference I can find to someone having success. My hardware is nRF52 based so I don't have the hardware to explicitly try this repo myself. I have tried various HID report map structures, from cloning existing ones from Xbox One and DS4, to minimal two-axis & two-button ones. None of these appear as controllers in the Game Controller framework on iOS or macOS.

The closest I've got is by using the PID and VID of the latest xbox one controller, which is also BLE. That does then show up in the game controller settings page, and is reported as a controller by the Game Controller framework, but I couldn't get the inputs to trigger any actions. I'd also much prefer not having to use PID/VID from other controllers.

Basically I'm just looking for a bit more information about what you've managed to get working on iOS and confirmation that it is the gamepad HID reports that are being exposed through Game Controller to iOS.

Thanks!

LeeNX, 2025-01-16:
@tangobravo I believe that iOS only supports [Mfi](https://mfi.apple.com) devices, which needs to be licensed. Thou I see that XBox and DualSense support might have been added since around iOS 14.5, but I can't say how well that has been tested. 
  
If you were able to get an iOS device to pick up your bluetooth devie with the xbox VID/PID, I am guessing Apple have added XInput support, which I would then suggest to look at the [ESP32-BLE-CompositeHID](https://github.com/Mystfit/ESP32-BLE-CompositeHID) project, as it has all the XInput stuff all collected together. Not sure if the ```ESP32-BLE-CompositeHID``` library will work with nRF52 device, but I have seen that [NimBLE-Arudino ](https://github.com/h2zero/NimBLE-Arduino) supports nRF52 devices. Would be interesting to seen.
  
But, I can confirm that on MacOS for the last three major releases support BLE Gamepad devices, even ones created by ESP32-BLE-Gamepad.

LeeNX, 2025-01-16:
Disclaimer, I don't have any iOS devices to report on, only OSX and Linux devices.

> There's a little more information here: https://forums.developer.apple.com/forums/thread/763679
> 
> MFi controllers were historically the only thing supported, and are "Classic Bluetooth" (ie not BLE). Support has definitely been widened since then, and at least one HID-over-BLE controller is officially supported as mentioned by the Apple engineer in that forum thread. Just no information about whether generic controllers are supposed to be supported or not.
> 
You really at the mercy of the developers of the Host OS and how much or how well they implement support for BLE devices. Been possible really does not explain what is support or how it works. I hope the more BLE gamepad devices are out there, the more BLE gamepad support will be implemented.

> > If you were able to get an iOS device to pick up your bluetooth devie with the xbox VID/PID, I am guessing Apple have added XInput support, which I would then suggest to look at the [ESP32-BLE-CompositeHID](https://github.com/Mystfit/ESP32-BLE-CompositeHID) project, as it has all the XInput stuff all collected together. Not sure if the `ESP32-BLE-CompositeHID` library will work with nRF52 device, but I have seen that [NimBLE-Arudino ](https://github.com/h2zero/NimBLE-Arduino) supports nRF52 devices. Would be interesting to seen.
> 
> Thanks for the link. From a quick glance it does look like that repo is also using the xbox VID and PID, they do appear in the XboxDescriptors.h file at least. Looks like that might be a requirement to get iOS to do anything with them.
> 
Yip, ESP32-BLE-CompositeHID is emulating or duplicating the XInput, which I believe is only supported on the xbox VID/PID. I believe any 3rd-party XBox controllers will have the same VID/PID as part of licensing the XBox tooling, but I don't have any other XBox device to confirm more than what I have seen online.

> > But, I can confirm that on MacOS for the last three major releases support BLE Gamepad devices, even ones created by ESP32-BLE-Gamepad.
> 
> Do you know if they are reported through the Game Controller framework, or through a different HID device API? I've been able to get a BLE HID device with a generic VID/PID to show up in System Settings -> Game Controllers on Mac but they are still not reported through the Game Controller framework.

I don't remember a ```System Settings -> Game Controllers```, I see a ```System Settings -> Game Center``` and mine is disabled. I do remember that I saw the device in ```System Settings -> Bluetooth``` and was able to use in Chrome Browser at https://hardwaretester.com/gamepad, thou I am now thinking this might have been using HIDAPI interface, but I would need to flash another test device to check. Do you have another OSX app that could be used to test the interface?

tangobravo, 2025-01-16:
Thanks for the responses, I'll compare against the ESP32-BLE-CompositeHID library.

The generic one does show up in hardwaretester.com/gamepad with Chrome on macOS, but not through the Game Controller framework. So Chrome likely has its own lower-level HID device implementation. The sample app I've been using on iOS and macOS for the Game Controller framework is here btw:
https://developer.apple.com/documentation/gamecontroller/supporting-game-controllers

We already have controllers out in the wild for our phone-based VR / MR setup. They currently use a simple BLE protocol rather than HID (documented [here](https://github.com/zappar-xr/zapbox-controller-protocol)) but I'm hoping with a firmware update I can broaden support to any other iOS apps with game controller support.