lemmingDev/ESP32-BLE-Gamepad #249: No bluetooth device visible

borsti2000, 2024-12-25:
No bluetooth device is visible. Not on PC not on Mobile (Android).
Tested on serveral esp32 dev modules with several example of the project.
I try also other bluetooth codes to test. If i flash the module with other code, the dev-board appears (as other device). 
If I flash again an example of this libary, the name of the bluetooth device not change but it is visible as gamecontroller (paring not possible).

LeeNX, 2025-01-03:
Possible related to https://github.com/lemmingDev/ESP32-BLE-Gamepad/pull/250. 
  
Would be up for testing master branch which this was merged into and letting us know if that solved your problem?

borsti2000, 2025-01-04:
Yes, now the board is visilble ans pairing is possible. It seems that was the issue.

lemmingDev, 2025-01-06:
> Possible related to #250.
> 
> Would be up for testing master branch which this was merged into and letting us know if that solved your problem?

I released a new version with this fix

I moved to Japan a month ago and forgot to take any ESP32 with me

I just ordered a regular ESP32, an ESP32-C3 and an ESP32-S3 from AliExpress, so I'll be able to actually test some stuff in a couple of weeks when they arrive

LeeNX, 2025-01-06:
Thanks @lemmingDev and enjoy the new adventure.  
  
@borsti2000 if this is resolved, mind closing the issue?

lemmingDev, 2025-01-12:
Guess it's resolved

Will close

closed 2025-01-12 by lemmingDev