lemmingDev/ESP32-BLE-Gamepad #251: Input is only recognized when first paired

Sflofler, 2025-01-03:
Well, like the title says, input is only registered when first paired with ESP32. I'm using windows joypad/game controllers to debug if an input is pressed, also tested with HTML gamepad tester btw.

I also changed the debug level for the NIMBLE library to get almost everything that I could while testing it and it actually behaves differently when first paired and when I restart the ESP32.

It is important to mention that the device re-pairs with windows, it says connected! But it doesn't register any input!

This is the log I get when I first pair with windows:
```
D NimBLEServer: Gatt Read event
D NimBLEServer: >> handleGapEvent:
D NimBLECharacteristicCallbacks: onStatus: default
D NimBLEServer: << handleGapEvent
``` 

And I don't get it when I restart the ESP32.

I've seen people complain about the same thing but the majority of the issues/posts on the internet was closed because it doesn't happen with everyone, by the looks of it.

Does the log give any clue about this issue? I'm currently trying to deeply debug it but I thought I should also open an issue during the proccess!

Thanks!

LeeNX, 2025-01-03:
You have not listed your sketch nor versions of used libraries, so I am making alot of guesses. ESP32-BLE-Gamepad v6.1 and NimBLE-Arduino v2.1.2?

Have you tried any of the ESP32-BLE-Gamepad example sketches? Do they work?

Next, if you make any changes to your sketches, I would always suggest deleting/unpairing host OS from the device, as changes to the setup are not changed on the host OS due to attributes been cached on first pairing.  
  
I would also enable serial debugging, where the button inputs are also echoed to the serial console, just to make sure your logic and hardware is correct and triggering.

Lastly, have you tried on a different host OS device, like Android device, where the bluetooth hardware and stack generally are a little less fluid or borkable?  
  
Looking for as much info that could help, do any of the other Bluetooth example sketches work for ESP32? Have you done debugging with something like Punchthrough's LightBlue?

Sflofler, 2025-01-03:
Sorry for not providing other infos!

Here is the code for testing that I'm currently running:
```c++
#include <Arduino.h>
#include <BleGamepad.h>

BleGamepad bleGamepad;
BleGamepadConfiguration bleGamepadConfig;

void setup()
{
    Serial.begin(115200);

    bleGamepadConfig.setAutoReport(false);
    bleGamepadConfig.setControllerType(CONTROLLER_TYPE_GAMEPAD);
    bleGamepadConfig.setButtonCount(3);

    bleGamepad.begin(&bleGamepadConfig);
    delay(3000);
}

void loop()
{
    if (bleGamepad.isConnected())
    {
        bleGamepad.press(BUTTON_1);
        bleGamepad.sendReport();
        delay(1000);
        bleGamepad.release(BUTTON_1);
        bleGamepad.sendReport();
        delay(1000);
    }
}
``` 

I'm running it on PlatformIO, the library version is 0.6.1 and NimBLE is 2.1.2.
```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
monitor_speed = 115200
lib_deps = lemmingdev/ESP32-BLE-Gamepad@^0.6.1
``` 

Yes I have tried the examples, they do work but I still get the connection issue.

On Android I just tried to connect the device and it said that the device connected needed an app.

No I didn't use Punchthrough's LightBlue or anything similar, just Windows native "gamepad debugger" and the HMLT gamepad testes, it's nice to mention that it works on both but only when first paired.

Oh and yes, I do unpair and pair it back when I change the code!
But if I don't change the code, disconnect the ESP32 from my pc and connect it back on, just for power, the bluetooth connection goes fine but the input recognition doesn't work anymore.

I actually just downloaded ArduinoIDE to see if I get any lucky since there were people with the same problem but reinstalling the IDE fixed the issue. Will test it soon!

LeeNX, 2025-01-03:
I have not tested your sketch, but two things stand out, first, why the long delay in the setup? Maybe you not coming out of the setup for the button actions to happen? I have never needed to add a delay in the setup.  
  
Second, your PlatformIO seems to be missing the NimBLE  library, this might just be a typo, this is my last test looks like https://gitlab.com/leet/ble-gamepad-collection/-/tree/main/platformio/ESP32-BLE-Gamepad-test-a?ref_type=heads. 
  
I would suggest using more serial output text for debugging, so that you know that you are actually know that the button actions are meant to be running.  
  
Very odd that it works the first time and not after. Not had this problem myself.  
  
I was able to use the [HTML Gamepad Test](https://hardwaretester.com/gamepad) on my Android device with my test setups.  

BLE-Mouse and/or BLE-keyboard both work without issues?  

Not sure what else to suggest.

Sflofler, 2025-01-03:
Well the NimBLE is not listed in the platformio.ini since BLEGamepad is a fork of it, isn't it? I can see the library being used with BLEGamepad without any issue, can compile and everything, even see and connect the device so I guess that not being listed there is not the problem.

I just connected it to my android device and went to the HTML Gamepad Test and it always works, I unplug the ESP32 from my pc and plug it back on, it auto connects to my android device and the website is showing the input being pressed as it should.

Well, I guess it is an issue with my BT dongle?

Sflofler, 2025-01-03:
Hey! So I just tested with BLE-Keyboard and it just works. I see in the library that NimBLE is disabled by default.

Maybe something with NimBLE? Is it possible to use BLE-Gamepad without NimBLE?

LeeNX, 2025-01-04:
Hey @Sflofler , I have an issue open and even worked on a PR to have access to both BLE stacks - https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/142. 
  
But I have not got it to work nor do I think there is enough interested to use ESP32-BLE-Gamepad without NimBLE-Arduino. Plus I think ```Espressif Systems``` is moving toward using NimBLE-Arduino as the default stack (thou I might be wrong on all cases).  
  
I have two other suggestions I did think of last night. If you just looking to get your project possible working and moving forward, maybe look [ESP32-BLE-CompositeHID](https://github.com/Mystfit/ESP32-BLE-CompositeHID), which is basically making an XInput device. My thinking is that XInput device like the XBox Gamepad is way more tested than BLE Gamepad as there are very few BLE Gamepad devices, unlike BLE keyboards & mice.  
  
The other option, at least from testing, debugging and trying to track down what could be the problem, if you have another ESP32 module, could setup a [Bluepad32](https://github.com/ricardoquesada/bluepad32) Host device. This is alot more work and I think it might not be a path any might do, but could be the start of a 3rd-Party USB Bluetooth dongle for input devices only 🤔

RaazP, 2025-01-05:
Hi everyone,
I somehow didn't see this issue and LeeNX kindly linked it to me.
Nothing to help, just stating that I'm having exactly the same issue.
My workaround for now is:
1. Windows -> forget device
2. ESP32 -> Reset
3. Windows -> search BT devices and connect

It's quickly done, just a few clicks and moving the EN Pin to the surface of my buttonbox with a small microswitch isn't an issue.

The bigger issue is that my racing sims detect is as a new device every time, so I have to re-map all 20 buttons before driving.

LeeNX, 2025-01-05:
Thanks @RaazP , not ideal, but is a work around, especially if you don't have resources to debug the problem.  
  
I have not seen this on my Macs nor my RetroPies, thou I have not done lots of testing with this problem. Next set of things I can think of, is listing OS and version, Bluetooth hardware and version and possible driver release for Bluetooth hardware? Basically need to find out what is the common effected setup, then we might be able to track down the problem or hopefully see some sort of commonality.

RaazP, 2025-01-05:
> Thanks @RaazP , not ideal, but is a work around, especially if you don't have resources to debug the problem.
>
> I have not seen this on my Macs nor my RetroPies, thou I have not done lots of testing [...]

I've tested a bit more and the good news: It's my code!
The bad news: It's my code...

When using the SingleButton.ino from the examples, all my ESP32 boards work flawlessly. Both on my Surface Pro 8 and my Desktop PC. Pressing reset, unplugging etc. no issues at all, happily reconnecting and pressing button 1 on pin 13.

So apparently it has nothing to do with hardware, Windows or drivers.
I've also put the boards on my breadboard with everything connected, but still using the SingleButton example and it works fine, so it also doesn't seem to be an electric problem.

I've attached my code, maybe you can easily spot an issue? I'm still quite a noob at coding, so it's 50% understanding what I did, 25% stitching together code from examples etc. and 25% writing my own code.

I would understand, if my code wouldn't work at all, but I really can't find anything in it, that would cause issues on re-connecting.

Thanks in advance!

`#include <Arduino.h>
#include <Keypad.h>     // https://github.com/Chris--A/Keypad
#include <BleGamepad.h> // https://github.com/lemmingDev/ESP32-BLE-Gamepad

BleGamepad bleGamepad("WirelessButtonplate", "Rasmus"); // Shows how you can customise the device name, manufacturer name and initial battery level

// Keypad Definitions-----------------------------------------------------------------------------
// Buttonplate Matrix
#define ROWS 3
#define COLS 4
uint8_t rowPins[ROWS] = {12, 14, 27}; // ESP32 pins used for rows      --> adjust to suit --> Pinout on board: R1, R2, R3, R4
uint8_t colPins[COLS] = {26, 25, 33, 32}; // ESP32 pins used for columns   --> adjust to suit --> Pinout on board: Q1, Q2, Q3, Q4
uint8_t keymap[ROWS][COLS] =
    {
        {1, 2, 3, 4},    // Buttons  1,  2,  3,  4      --> Used for calulating the bitmask for sending to the library
        {5, 6, 7, 8},    // Buttons  5,  6,  7,  8      --> Adjust to suit which buttons you want the library to send
        {9, 10, 11, 12} // Buttons  9, 10, 11, 12      -->
        //{13, 14, 15, 16} // Buttons 13, 14, 15, 16      --> Eg. The value 12 in the array refers to button 12
};

Keypad customKeypad = Keypad(makeKeymap(keymap), rowPins, colPins, ROWS, COLS);

// Single Buttons Definitions--------------------------------------------------------------------
// Hornbutton and 7-Way-Push
#define numOfButtons 2 // Number of buttons here
byte previousButtonStates[numOfButtons];  // ignore
byte currentButtonStates[numOfButtons];   // ignore
// Pins Numbers (GPIO** from PinOut)
byte buttonPins[numOfButtons] = {13, 23};
// Number in Windows for each pin
byte physicalButtons[numOfButtons] = {13, 14};

// POV Hat Definitions---------------------------------------------------------------------------
// 7-Way R/L/Up/Down
#define numOfHats 1     // Number of DPADs here
byte previousHatStates[numOfHats * 4];  // ignore
byte currentHatStates[numOfHats * 4];   // ignore
byte hatPins[numOfHats * 4] = {21, 3, 1, 22}; // in order UP, LEFT, DOWN, RIGHT. GPIO Pin-numbers here

// CTS288 Definitions
byte previousCTS288_1;
byte currentCTS288_1;
byte cts_1_Pin_A = 5;
byte cts_1_Pin_B = 17;
byte cts_1_Button_A = 15;
byte cts_1_Button_B = 16;
bool cts_1_Pressed = false;
unsigned long cts_1_LastPressTime = 0;

// 7-Way-Encoder Definitions
#define inputCLK 18 //(A)
#define inputDT 19  //(B)
[... 199 lines cut ...]

  for (int i = 0; i < LIST_MAX; i++) // Scan the whole key list.      //LIST_MAX is provided by the Keypad library and gives the number of buttons of the Keypad instance
  {
    if (customKeypad.key[i].stateChanged) // Only find keys that have changed state.
    {
      uint8_t keystate = customKeypad.key[i].kstate;

      if (bleGamepad.isConnected())
      {
        if (keystate == PRESSED)
        {
          bleGamepad.press(customKeypad.key[i].kchar);
        } // Press or release button based on the current state
        if (keystate == RELEASED)
        {
          bleGamepad.release(customKeypad.key[i].kchar);
        }

        //bleGamepad.sendReport(); // Send the HID report after values for all button states are updated, and at least one button state had changed
      }
    }
  }
}

`

travisrroy, 2025-01-07:
I am experiencing the same thing where after a disconnect, I have to go through the entire re-pairing process to get my buttons to work again. Tried downgrading to 0.5.4 and 0.5.6 with NimBLE 1.41 but my code doesn't seem to be compatible. 

NimBLE 2.1.3 was just released, but I see the same issue with 0.6.2.

LeeNX, 2025-01-07:
@travisrroy , if you have a second ESP32, why not try [Bluepad32](https://github.com/ricardoquesada/bluepad32) as a host and see if that has the same problem with your project.  
  
Else test with a mobile device like android. Or Install [Lightblue](https://punchthrough.com/lightblue/).  
  
Another idea, enable the battery level ramping and rolling, see if you can see that, even if the input does not work. Serial debugging, to confirm input should be firing too.  
  
I could make Bluepad32 basic firmwares for ESP32 and ESP32-C3, for testing has hosts. Not sure what else might be useful for testing and debugging?

uggima, 2025-01-07:
> > Thanks @RaazP , not ideal, but is a work around, especially if you don't have resources to debug the problem.
> > I have not seen this on my Macs nor my RetroPies, thou I have not done lots of testing with this problem. Next set of things I can think of, is listing OS and version, Bluetooth hardware and version and possible driver release for Bluetooth hardware? Basically need to find out what is the common effected setup, then we might be able to track down the problem or hopefully see some sort of commonality.
> 
> I've tested a bit more and the good news: It's the code! The bad news: It's the code...
> 
> **Update 1:** The SingleButton.ino works fine, the MultipleButtons.ino example doesn't and shows the same re-connect issue, so I'm gonna try to adjust the MultipleButtons example until it works.
> 
> **Update 2:** I deleted my long custom sketch and found out, that it might be the combination of bleGamepadConfig.setAutoReport(false); and bleGamepad.sendReport();
> 
> When I put them into the SingleButton.ino, the re-connect issue happens.
> 
> **Update 3:** The AutoReport and sendReport break the SingleButton sketch, but the issue still happens, when I disable the two lines in the MultipleButtons sketch. I've attached the MultipleButton code at the end, so you don't have to open the examples yourself.
> 
> **Update 4:** The "MultipleButtonsDebounce" example works flawlessly. Although setAutoReport(false) and sendReport() are used. However, making it a 22 Button Sketch (all Pins with in-built Pull-ups) as individual buttons breaks it. I can set "setButtonCount(22)" though, without configuring the pins & buttons. I then see 22 buttons in Windows, but of course only 10 of them work. No re-connect issue. (Important in Update 5)
> 
> Sadly, my skills aren't there yet to find out the difference and why the debounce example with 10 buttons works, while the normal one doesn't. And also why the Denounce example breaks with 22 configured buttons.. But there's the path to follow, if someone has the time and energy to work it out. Sadly, I'm not sure if I can manage to adjust the hat-switch, button-matrix (keyboard.h) and rotary encoder codes into the "debounce-style". (Update 5 finds out, that I don't have to do that)
> 
> **Update 5:** Now it goes really weird! I can flash the "MultipleButtonsDebounce" with "setButtonCount(22)", but only configure 10 pins. Then connect it to my PC and it works perfectly. I can then flash my buttonplate sketch with the 3x4-button-matrix, 6 buttons and two rotary encoders (essentially just 22 buttons, with a statemachine for encoders) just fine, without going to the BT settings and doing the usual forget device -> add device. If I do that, it breaks. I even unplugged the ESP32, restarted the PC, plugged-in the ESP32 and it works flawlessly.
> 
> So I basically tricked the "bug" and would be completely fine with this, but I'd really like to get 4 of the buttons registered as "Hat", not as button. Using the hat functions in my buttonplate sketch doesn't work, since the MultipleButtonsDebounce sketch doesn't set any hats. If I add "bleGamepadConfig.setHatSwitchCount(1);" to the MultipleButtonsDebounce sketch, it breaks on re-connect.
> 
> **Conclusion for now:**
> 
>     1. Flash with MultipleButtonsDebounce, but setButtonCount(22), instead of (10). Or however as many buttons you need.
>        But don't change anything else!
> 
>     2. Add device in Windows BT settings
> 
>     3. Flash with custom sketch, which contains the actual code for all the buttons.
> 
>     4. DON'T forget+add device! Just let it re-connect automatically after flashing.
> 
>     5. Windows now thinks it's still the MultipleButtonsDebounce device, but all buttons (22 for me) work as they should and re-connecting works like a charm, even after restarting the PC, disabling BT etc.
> 
> 
> **Original post:** When using the SingleButton.ino from the examples, all my ESP32 boards work flawlessly. Both on my Surface Pro 8 and my Desktop PC. Pressing reset, unplugging etc. no issues at all, happily reconnecting and pressing button 1 on pin 13.
> 
> So apparently it has nothing to do with hardware, Windows or drivers. I've also put the boards on my breadboard with everything connected, but still using the SingleButton example and it works fine, so it also doesn't seem to be an electric problem.
> 
> I'm still quite a noob at coding, so it's 50% understanding what I did, 25% stitching together code from examples etc. and 25% writing my own code. I would understand, if my code wouldn't work at all, but I really can't find anything in it, that would cause issues on re-connecting.
> 
> Thanks in advance!
> 
> This works fine:
> 
> ```c++
> /*
>  * This code programs a number of pins on an ESP32 as buttons on a BLE gamepad
>  *
>  * It uses arrays to cut down on code
>  *
>  * Uses the Bounce2 library to debounce all buttons
>  *
>  * Uses the rose/fell states of the Bounce instance to track button states
>  *
>  * Before using, adjust the numOfButtons, buttonPins and physicalButtons to suit your senario
>  *
>  */
> 
> #define BOUNCE_WITH_PROMPT_DETECTION // Make button state changes available immediately
> 
> #include <Arduino.h>
> #include <Bounce2.h>    // https://github.com/thomasfredericks/Bounce2
> #include <BleGamepad.h> // https://github.com/lemmingDev/ESP32-BLE-Gamepad
> 
> #define numOfButtons 10
> 
> Bounce debouncers[numOfButtons];
> BleGamepad bleGamepad;
> 
[... 115 lines cut ...]
[... cut ...]
>         if (currentButtonStates != previousButtonStates)
>         {
>             for (byte currentIndex = 0; currentIndex < numOfButtons; currentIndex++)
>             {
>                 previousButtonStates[currentIndex] = currentButtonStates[currentIndex];
>             }
> 
>             //bleGamepad.sendReport();
>         }
> 
>         delay(20);
>     }
> }
> ```

Tested working, (does not press buttons automatically you have to ground one of "buttonPins") maybe the Autoreply fails to actually update anything past initial connection?

RaazP, 2025-01-08:
Thank you all for contributing!
I've enabled these in the nimconfig.h:

#define CONFIG_BT_NIMBLE_LOG_LEVEL 1
#define CONFIG_NIMBLE_CPP_LOG_LEVEL 3
#define CONFIG_NIMBLE_CPP_DEBUG_ASSERT_ENABLED 1
#define CONFIG_NIMBLE_CPP_ENABLE_RETURN_CODE_TEXT
#define CONFIG_NIMBLE_CPP_ENABLE_GAP_EVENT_CODE_TEXT
#define CONFIG_NIMBLE_CPP_ENABLE_ADVERTISEMENT_TYPE_TEXT

This is what I did and what shows the issue:
```C++
// forgot device
// flashed
// opened serial monitor in Arduino IDE

// reset button
// BT device search & connected
// buttons pressed and working in-game

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4688
load:0x40078000,len:15460
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3196
entry 0x400805a4
I NimBLEDevice: BLE Host Task Started
I NimBLEDevice: NimBle host synced.
I NimBLEServer: mtu update event; conn_handle=0 mtu=255
I NimBLEServer: subscribe event; attr_handle=8, subscribed: true
I NimBLEServer: subscribe event; attr_handle=44, subscribed: true
I NimBLEServer: subscribe event; attr_handle=39, subscribed: true
ALPS 5 pushed.
ALPS 4 pushed.
ALPS 3 pushed.
ALPS 1 pushed.
ALPS 2 pushed.
ALPS_Encoder_B_Pressed
ALPS_Encoder_A_Pressed

// reset button
// auto re-connected
// buttons pressed, NOT working in-game
// but code is working, serial output shows them fine

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4688
load:0x40078000,len:15460
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3196
entry 0x400805a4
I NimBLEDevice: BLE Host Task Started
I NimBLEDevice: NimBle host synced.
I NimBLEServer: subscribe event; attr_handle=8, subscribed: true
I NimBLEServer: subscribe event; attr_handle=44, subscribed: true
I NimBLEServer: mtu update event; conn_handle=0 mtu=255
ALPS 5 pushed.
ALPS 4 pushed.
ALPS 3 pushed.
ALPS 1 pushed.
ALPS 2 pushed.
ALPS_Encoder_B_Pressed
ALPS_Encoder_A_Pressed```

If I first flash the standard MultipleButtonsDebounce.ino with numOfButtons = 22, the two blocks in the serial monitor are identical:
```C++
// forgot device
// flashed MultipleButtonsDebounce.ino with numOfButtons = 22;
// reset button
// BT connected
// buttons pressed and working

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4688
load:0x40078000,len:15460
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3196
entry 0x400805a4
I NimBLEDevice: BLE Host Task Started
I NimBLEDevice: NimBle host synced.
I NimBLEServer: mtu update event; conn_handle=0 mtu=255
I NimBLEServer: subscribe event; attr_handle=8, subscribed: true
I NimBLEServer: subscribe event; attr_handle=44, subscribed: true
I NimBLEServer: subscribe event; attr_handle=39, subscribed: true
Button 6 pushed.
Button 6 released.
Button 6 pushed.
Button 6 released.

// reset button
// auto re-connected
// buttons pressed and working

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4688
load:0x40078000,len:15460
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3196
entry 0x400805a4
I NimBLEDevice: BLE Host Task Started
I NimBLEDevice: NimBle host synced.
I NimBLEServer: subscribe event; attr_handle=8, subscribed: true
I NimBLEServer: subscribe event; attr_handle=44, subscribed: true
I NimBLEServer: subscribe event; attr_handle=39, subscribed: true
I NimBLEServer: mtu update event; conn_handle=0 mtu=255
Button 6 pushed.
Button 6 released.
Button 6 pushed.
Button 6 released.
```

Sflofler, 2025-01-09:
I teste the `SingleButton` example, straight copied it and pasted and It worked but when I added another button in the exact same way that is shown in the example It stopped working.

LeeNX, 2025-01-09:
@Sflofler did you removed /deleted /unpaired the device on the host OS and try
re-pairing between changes?

Host OS are known to cache the ble setup and changes you make while
testing, like new buttons do not get picked up and even break device usage.

Sflofler, 2025-01-09:
@LeeNX Yes, everytime I upload a new code I do remove the paired device from Windows Bluetooth Devices!

LeeNX, 2025-01-09:
Thanks @Sflofler. Last suggestion if you willing to test for today, how about switching between two know working sketches, just to rule out some typo or IDE/editor issue. Try the ```FlightControllerTest.ino``` from the examples and see if that breaks. It also has a different device name, which could be something to look into.  
  
Best of luck.

lemmingDev, 2025-01-09:
So, I just had some ESP32 arrive and did some testing

Both on ESP32 and ESP32-C3 -> controls work on first upload, remove device and it reconnects, but controls no longer work

lemmingDev, 2025-01-09:
Yeah - Windows

Just tested, and reconnections to Android seem fine, and functionality persists

May be an issue with Windows compatibility. I wonder what's changed in either NimBLE or Windows?

Sflofler, 2025-01-09:
I had no issues on Android with the first sketch that I uploaded in this issue, which is this one:

```c++
#include <Arduino.h>
#include <BleGamepad.h>

BleGamepad bleGamepad;
BleGamepadConfiguration bleGamepadConfig;

void setup()
{
    Serial.begin(115200);

    bleGamepadConfig.setAutoReport(false);
    bleGamepadConfig.setControllerType(CONTROLLER_TYPE_GAMEPAD);
    bleGamepadConfig.setButtonCount(3);

    bleGamepad.begin(&bleGamepadConfig);
    delay(3000);
}

void loop()
{
    if (bleGamepad.isConnected())
    {
        bleGamepad.press(BUTTON_1);
        bleGamepad.sendReport();
        delay(1000);
        bleGamepad.release(BUTTON_1);
        bleGamepad.sendReport();
        delay(1000);
    }
}
``` 

Reconnection worked everytime and button press was also fine!

If I had to take a guess I would say it's something with NimBLE + Windows becaus when I tested [BLE-Keyboard](https://github.com/T-vK/ESP32-BLE-Keyboard) **without activating NimBLE** It worked! I don't have the sketch anymore but It was also very simple, I used the touch sensor on the ESP32 to open the Calculator on Windows.

h2zero, 2025-01-11:
Hello everyone, I have identified the issue and provided a potential solution #257, the error occurs in the NimBLE host stack and therefore would need resolution upstream from my library, which I will pursue later. For now the above PR should be a satisfactory workaround, please let me know if it works for you or not.

closed 2025-01-12 by lemmingDev

lemmingDev, 2025-01-12:
Tested and working

Thanks so much!

Sflofler, 2025-01-12:
Yup! Can confirm that it is working!

Again with this sketch:
```c++
#include <Arduino.h>
#include <BleGamepad.h>

BleGamepad bleGamepad;
BleGamepadConfiguration bleGamepadConfig;

void setup()
{
    Serial.begin(115200);

    bleGamepadConfig.setAutoReport(false);
    bleGamepadConfig.setControllerType(CONTROLLER_TYPE_GAMEPAD);
    bleGamepadConfig.setButtonCount(3);

    bleGamepad.begin(&bleGamepadConfig);
    delay(3000);
}

void loop()
{
    if (bleGamepad.isConnected())
    {
        bleGamepad.press(BUTTON_1);
        bleGamepad.sendReport();
        delay(1000);
        bleGamepad.release(BUTTON_1);
        bleGamepad.sendReport();
        delay(1000);
    }
}
``` 

Thank you!

PR #257 (2025-01-12): Fix Windows notifications not being sent on reconnection.
When Windows bonds with a device and subscribes to notifications/indications of the device characteristics it does not re-subscribe on subsequent connections. If a notification is sent when Windows reconnects it will overwrite the stored subscription value in the NimBLE stack configuration with an invalid value which results in notifications/indications not being sent.

This change will ensure that no notification or indication is sent before authentication upon reconnection, avoiding the described issue described above.

Fixes #251